78/xiaozhi-esp32 #324: 关于OTA

49949046, 2025-03-11:
### Is your feature request related to a problem?

源码修改OTA地址编译后，服务器端更新OTA流程和代码是否可以开源？

78, 2025-03-11:
你好，目前没有开源计划。
请参考OTA接口的文档自行实现：https://ccnphfhqs21z.feishu.cn/wiki/FjW6wZmisimNBBkov6OcmfvknVd

guoguoguo111, 2025-03-17:
> 你好，目前没有开源计划。 请参考OTA接口的文档自行实现：https://ccnphfhqs21z.feishu.cn/wiki/FjW6wZmisimNBBkov6OcmfvknVd
这个文档不适用于现在的代码了吧，我看现在大致如下呢：
Device-Id：
Accept-Language：
User-Agent board /
    /* 
        {
            "version": 2,
            "flash_size": 4194304,
            "psram_size": 0,
            "minimum_free_heap_size": 123456,
            "mac_address": "00:00:00:00:00:00",
            "uuid": "00000000-0000-0000-0000-000000000000",
            "chip_model_name": "esp32s3",
            "chip_info": {
                "model": 1,
                "cores": 2,
                "revision": 0,
                "features": 0
            },
            "application": {
                "name": "my-app",
                "version": "1.0.0",
                "compile_time": "2021-01-01T00:00:00Z"
                "idf_version": "4.2-dev"
                "elf_sha256": ""
            },
            "partition_table": [
                "app": {
                    "label": "app",
                    "type": 1,
                    "subtype": 2,
                    "address": 0x10000,
                    "size": 0x100000
                }
            ],
            "ota": {
                "label": "ota_0"
            },
            "board": {
                ...
            }
        }
    */

78, 2025-03-17:
文档还是准确的，服务器会根据开发板的类型，分区空间，固件版本号，芯片类型，flash大小，是否有psram等综合因素，以及用户是否在后台打开了OTA升级，来决定是否返回新版本的固件。

closed 2025-03-21 by 78