78/xiaozhi-esp32 #327: build flash 成功，但是屏幕不亮

shenxiaolong, 2025-03-11:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

1.4.6

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

虾哥好，很开心能遇到 xiaozhi 这么棒的产品。

我 git clone 代码后，执行到 idf.py build flash，都没有问题，但是屏幕不亮，我的屏幕是 OLED 0.91。
我在 idf.py menuconfig 中对 Board Type 选择的是 面包板新版接线（WiFi）+ LCD，是不是我这里的 Board Type 选错了？

[image]

我可以确保线连接的是没有问题的，因为我烧录某QQ群里下载的 merged-binary.bin 文件屏幕是可以正常使用的。

### What is the actual behavior?

none

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

zactang, 2025-03-13:
我用最新代码执行跟楼上的情况差不多：我提示zip相关错误。但是看到说可以忽略不影响。但实际情况，我这边烧录完成后也是屏幕不亮，又刷回了原来的bin文件正常。

[image]

[image]

zactang, 2025-03-13:
board type 设置为 ： 面包板新版接线（WiFi）。试下

closed 2025-03-21 by 78