78/xiaozhi-esp32 #331: v1.4.6_magiclick几个版本什么意思？

liuhedev, 2025-03-12:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.4.6_magiclick-2p4

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

[image]

### What is the actual behavior?

[image] 这几个版本什么区别？

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

78, 2025-03-12:
@MakerM0 作者来解释一下？

MakerM0, 2025-03-23:
对应不同版本的硬件，不能互刷

closed 2025-05-20 by 78