78/xiaozhi-esp32 #337: Speaking 状态下 接收服务器回传数据异常时无法离开 speaking 状态

laride, 2025-03-13:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.4.6

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

在 speaking 状态下触发 `Application::ToggleChatState()` 或 `Application::AbortSpeaking(AbortReason)`，对话应当立刻切换至 listening 状态。

### What is the actual behavior?

listening 状态结束、进入 speaking 状态时，设备可能出现以下两种情况
- 不打印 `Application: << 回传文字内容` log
- 打印 `Application: << 回传文字内容` log，但没有音频播放

此时，设备卡在 speaking 状态，无法自动恢复正常；手动触发 `Application::AbortSpeaking(AbortReason)` 无法离开 speaking 状态，只能通过重启/reset恢复。

### Steps to reproduce.

普通对话时可能触发

### Debug Logs.

```plain
I (75719) Application: STATE: listening
I (80507) Application: >> 给我说个笑话吧
I (80529) Application: STATE: speaking
[触发 Application::AbortSpeaking(kAbortReasonNone)]
I (93183) Application: Abort speaking
[触发 Application::AbortSpeaking(kAbortReasonNone)]
I (97442) Application: Abort speaking
[触发 Application::AbortSpeaking(kAbortReasonNone)]
I (101643) Application: Abort speaking
```

78, 2025-03-13:
偶尔发生的吗？ 看起来是服务器没有返回tts.stop

WJ-JAY, 2025-03-13:
> 偶尔发生的吗？看起来是服务器没有返回tts.stop

我也发现了，概率蛮大,日志看也是服务器没有返回tts.stop，是否跟网络不稳定有关？

78, 2025-03-13:
> > 偶尔发生的吗？看起来是服务器没有返回tts.stop
> 
> 我也发现了，概率蛮大,日志看也是服务器没有返回tts.stop，是否跟网络不稳定有关？

Wi-Fi还是4G版本呢？

WJ-JAY, 2025-03-14:
> > > 偶尔发生的吗？看起来是服务器没有返回tts.stop
> > 
> > 我也发现了，概率蛮大,日志看也是服务器没有返回tts.stop，是否跟网络不稳定有关？
> 
> Wi-Fi还是4G版本呢？

WiFi

WJ-JAY, 2025-03-14:
> > > 偶尔发生的吗？看起来是服务器没有返回tts.stop
> > 
> > 
> > 我也发现了，概率蛮大,日志看也是服务器没有返回tts.stop，是否跟网络不稳定有关？
> 
> Wi-Fi还是4G版本呢？

经常发生在按键调用ToggleChatState打断对话之后

78, 2025-03-14:
> > > > 偶尔发生的吗？看起来是服务器没有返回tts.stop
> > > 
> > > 
> > > 我也发现了，概率蛮大,日志看也是服务器没有返回tts.stop，是否跟网络不稳定有关？
> > 
> > 
> > Wi-Fi还是4G版本呢？
> 
> 经常发生在按键调用ToggleChatState打断对话之后

有可能是最近更新的版本的bug，不过还需要研究如何快速重现。

manicwy, 2025-03-14:
> > > 偶尔发生的吗？看起来是服务器没有返回tts.stop
> > 
> > 
> > 我也发现了，概率蛮大,日志看也是服务器没有返回tts.stop，是否跟网络不稳定有关？
> 
> Wi-Fi还是4G版本呢？

我的4G版本，最近新版本也会出现这种情况，只能rst重启，感觉是服务器接口压力上来了？

78, 2025-03-14:
> > > > 偶尔发生的吗？看起来是服务器没有返回tts.stop
> > > 
> > > 
> > > 我也发现了，概率蛮大,日志看也是服务器没有返回tts.stop，是否跟网络不稳定有关？
> > 
> > 
> > Wi-Fi还是4G版本呢？
> 
> 我的4G版本，最近新版本也会出现这种情况，只能rst重启，感觉是服务器接口压力上来了？

设备IP是国内的吗？最近在升级海外加速线路，不知道跟这个有没有关系。

WJ-JAY, 2025-03-14:
> > > > > 偶尔发生的吗？看起来是服务器没有返回tts.stop
> > > > 
> > > > 
> > > > 我也发现了，概率蛮大,日志看也是服务器没有返回tts.stop，是否跟网络不稳定有关？
> > > 
> > > 
> > > Wi-Fi还是4G版本呢？
> > 
> > 
> > 我的4G版本，最近新版本也会出现这种情况，只能rst重启，感觉是服务器接口压力上来了？
> 
> 设备IP是国内的吗？最近在升级海外加速线路，不知道跟这个有没有关系。

我的是国内

78, 2025-03-14:
可以这么做，至于是两次还是三次强行进入idle，需要研究一下

HoGC, 2025-03-15:
应该是tts失败了，和音色有关系，可以更换湾湾小何看看

manicwy, 2025-03-16:
> > > > > 偶尔发生的吗？看起来是服务器没有返回tts.stop
> > > > 
> > > > 
> > > > 我也发现了，概率蛮大,日志看也是服务器没有返回tts.stop，是否跟网络不稳定有关？
> > > 
> > > 
> > > Wi-Fi还是4G版本呢？
> > 
> > 
> > 我的4G版本，最近新版本也会出现这种情况，只能rst重启，感觉是服务器接口压力上来了？
> 
> 设备IP是国内的吗？最近在升级海外加速线路，不知道跟这个有没有关系。

国内4G物联网

78, 2025-03-16:
确实是跟音色有关，应该是用的阿里云的音色，看到有不少报错，临时修改忽略了一下错误，不知道有没有改善（正确处理方式可能是重试）