78/xiaozhi-esp32 #343: 正在寻找可用的服务

zactang, 2025-03-14:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.4.7

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

我用的时websocket协议，我确保子的服务端是ok的。
但是板子连接上去提示：正在寻找可用的服务。  

我的板子相关配置：
          board type :  面包板新版接线（WiFi）
         Connection Type：Websocket

[image]

[image]

### What is the actual behavior?

1

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

78, 2025-03-14:
固件的通信协议选择了MQTT才会报这个错误？

zactang, 2025-03-14:
看来代码 确实 是MQTT 才会这样的，但是我选项确实 设置了socket
[image]

zactang, 2025-03-15:
问题：服务端防火墙没有关闭，所以本机测试ok。其他设备通讯进来有问题。

closed 2025-03-15 by zactang

iot-mcy, 2025-03-15:
ipv6连接也会提示“正在寻找可用的服务”
ipv4是可以的
防火墙已关
ipv6能ping通
[image]

78, 2025-03-15:
> ipv6连接也会提示“正在寻找可用的服务” ipv4是可以的 防火墙已关 ipv6能ping通 是服务端不支持ipv6监听吗
> 
> [image]

也许这个websocket的库，需要做一些配置