78/xiaozhi-esp32 #345: 今天突然出现这个问题：叫小智退下，它直接没声音了，然后半天也叫不答应

Ariken2002, 2025-03-14:
今天突然出现这个问题：叫小智退下，它直接没声音了，然后半天也叫不答应，我看后台的对话历史，叫了退出、退下这种词后，它回答是空白，没有对话内容，然后设备上就直接没声音，而且半天叫小智也没有反应了，要等一段时间叫才有反应，这是哪里的问题呢？之前一直正常，今天用的时候才出现的。

[image]

[image]

[image]

[image]

Ariken2002, 2025-03-14:
以前叫它退下，它会说一句话，然后就待命了，一切正常，但今天我叫了好多次，都是这个情况。

78, 2025-03-14:
> 以前叫它退下，它会说一句话，然后就待命了，一切正常，但今天我叫了好多次，都是这个情况。

有可能是llm，也可能是tts卡死了，需要调研。

Ariken2002, 2025-03-14:
> > 以前叫它退下，它会说一句话，然后就待命了，一切正常，但今天我叫了好多次，都是这个情况。
> 
> 有可能是llm，也可能是tts卡死了，需要调研。

出现的频繁还非常高，基本10次说退出，有9次都直接没反应了，不知道应该怎么处理了。

Ariken2002, 2025-03-14:
如果后台历史对话那里，是有文字的，它就能正常说出来，不然如果后台那里没有文字，是空白的，它就不会有声音，就一直没声了，叫也叫不了了。

78, 2025-03-14:
> 如果后台历史对话那里，是有文字的，它就能正常说出来，不然如果后台那里没有文字，是空白的，它就不会有声音，就一直没声了，叫也叫不了了。

你先备份一下记忆体，然后试试清空记忆体，会变好吗？不确定是否记忆体有字符导致大模型报错了，因为此前确实有用户遇到记忆里出现了不合适的内容，后台调用大模型的API时报错，会出现这种情况

Ariken2002, 2025-03-14:
> > 如果后台历史对话那里，是有文字的，它就能正常说出来，不然如果后台那里没有文字，是空白的，它就不会有声音，就一直没声了，叫也叫不了了。
> 
> 你先备份一下记忆体，然后试试清空记忆体，会变好吗？不确定是否记忆体有字符导致大模型报错了，因为此前确实有用户遇到记忆里出现了不合适的内容，后台调用大模型的API时报错，会出现这种情况

记忆体清除过几次，还是一样的问题，没有改善，我刷的是最新的1.4.6，之前的1.3.0好像没这个问题，不知道是不是新版本的原因，我重新刷回1.3.0看看，不知道为什么说退出时，ai的回答经常是空的，之前还从来没遇到过这种情况。

78, 2025-03-14:
可以对比一下不同版本。
少于3次对话，不会保存

Ariken2002, 2025-03-14:
> 可以对比一下不同版本。 少于3次对话，不会保存

不行，还是很频繁，一让退下，就直接显示“说话中”，然后就一直这样，也唤醒不了了，另外还经常出现图片下面的报错，一旦报错，声音就会有断断续续的情况，不知道是为什么（之前用DeepSeek V3模型时，这个错误出现得很频繁，后面又改回Qwen了，用Qwen前段时间很少出现，最近也出现得比较多了）：

[image]

Ariken2002, 2025-03-14:
> 可以对比一下不同版本。 少于3次对话，不会保存

让它退出经常卡住的这个问题着实有点烦，我直接改了代码，让它在说话的时候，也接收唤醒，临时解决一下这个烦人的问题。

Ariken2002, 2025-03-14:
看上去不是版本的问题，LLM问题的可能性更大，但找不到原因，为什么说退出，退下这类词，会返回空，想不太明白了。

78, 2025-03-14:
> 看上去不是版本的问题，LLM问题的可能性更大，但找不到原因，为什么说退出，退下这类词，会返回空，想不太明白了。

要不，你创建一个新的智能体，然后把设备解绑，使用新的智能体试试……

Ariken2002, 2025-03-15:
> > 看上去不是版本的问题，LLM问题的可能性更大，但找不到原因，为什么说退出，退下这类词，会返回空，想不太明白了。
> 
> 要不，你创建一个新的智能体，然后把设备解绑，使用新的智能体试试……

我先直接改一下提示词吧，让它在退出时说固定的词，先这样临时处理一下。

closed 2025-03-15 by Ariken2002