78/xiaozhi-esp32 #350: 行空板df-k10刷1.4.6版本固件,麦克风收音效果不好

nbzx, 2025-03-15:
唤醒词需要离麦克风很近才能唤醒
xiaozhi.me官网后台听历史对话的录音，声音很轻有杂音
我的板子是新的，自带固件远距离唤醒没有问题

78, 2025-03-15:
@HonestQiao 如果自带固件没问题，那么估计是AudioCodec的问题

nbzx, 2025-03-15:
感谢虾哥的及时回复
关于AudioCodec，有解决思路吗？

nbzx, 2025-03-16:
搜了一下，发现espressif官方论坛有帖子提到同样用es7243e录音芯片的esp32-s3-box-lite存在录音音量很小的问题，难道是驱动有问题？ [https://esp32.com/viewtopic.php?t=33759](url)

nbzx, 2025-03-16:
按照官方论坛的方法，修改db值为37.5，麦克风的确改善了很多
我不知道怎么传递get_db_reg函数的值，只能暴力修改es7243e.c测试了一下：
static uint8_t get_db_reg(float db)
{
    // reg: 12 - 34.5dB, 13 - 36dB, 14 - 37.5dB
    // db += 0.5;
    // if (db <= 33.0) {
    //     return (uint8_t) db / 3;
    // }
    // if (db < 36.0) {
    //     return 12;
    // }
    // if (db < 37.0) {
    //     return 13;
    // }
    return 14;
}

78, 2025-03-16:
按照代码，db传入只要大于等于37，都能得到14

mtdxc, 2025-03-17:
亲测在esp32s3-box-lite上也有效，之前为了快速移植，简单注释掉增益的代码，导致声音有点小。