78/xiaozhi-esp32 #351: 很恶心的BUG

CrisXie4, 2025-03-16:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.4.6_bread-compact-ml307

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

正常说话

### What is the actual behavior?

第一次叫他他有时候就没声音，只有字幕。
还有他会像这样说话：你你你你你你  

（重启）

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

CrisXie4, 2025-03-16:
这是在ESP32-S3 N16R8 上有的BUG

ForgetCSX, 2025-04-07:
我用的也是R16N8的模块啊，音频是8311和7210，没有遇到你说的这个，版本1.5.6

78, 2025-04-07:
后台语言偏好改为中文，或者不限定，试试。

closed 2025-05-20 by 78