78/xiaozhi-esp32 #369: iot定时控制设备，询问设备状态不一致

A29415, 2025-03-19:
### Is your feature request related to a problem?

请教下，在iot部分想实现一个定时控制设备的功能，例如告诉小智一分钟后打开风扇，主板A会通过串口发送指令给另一个主板B(该主板用于控制设备)，定时在主板B计时，定时结束后B会返回风扇状态给A，但是这时询问小智风扇状态，却还是关闭的，说明风扇状态没有同步。我想知道怎么修改能够同步这个状态？

HelloDB, 2025-03-19:
当你主板A收到B返回的状态值时，将这个值记录到你创建的那个iot的属性 `properties_.AddBooleanProperty()`里，这样你后面询问风扇是否开启 return 的就是最新状态值了。你举例的这个需求拿 `iot/things/lamp.h` 稍微改下就可以了，init 时创建一个管理A、B串口收发的线程和一个记录状态的变量，这个变量就是用来同步状态的。

A29415, 2025-03-19:
> 当你主板A收到B返回的状态值时，将这个值记录到你创建的那个iot的属性 `properties_.AddBooleanProperty()`里，这样你后面询问风扇是否开启 return 的就是最新状态值了。你举例的这个需求拿 `iot/things/lamp.h` 稍微改下就可以了，init 时创建一个管理A、B串口收发的线程和一个记录状态的变量，这个变量就是用来同步状态的。

请问是图片里的这个变量记录吗？我是使用类似的写法的，解析串口收到的指令并赋值给它，但是问小智并没有更新状态
[image]

HelloDB, 2025-03-19:
就是这个，你在return之前用printf打印出来看看是不是没赋值到这个变量。

78, 2025-03-19:
目前只有在状态变为Listening时，会调用UpdateIotStates发送变化的状态。

A29415, 2025-03-19:
> 根据你的需求，建议可以等MCP接口上线后会更灵活。

嗯嗯，感谢解惑，我明白问题所在了

HelloDB, 2025-03-22:
> 根据你的需求，建议可以等MCP接口上线后会更灵活。

大佬，MCP大概多久会上？现在让小智控制本地设备和播放本地歌曲都是在IOT属性里喂列表，用 `QwenMax2.5` 它能理解列表提取控制需要的参数，但目前这种方式已经到极限了，QW2.5 理解是最准的，成功率也只能95左右。

顺便问下 `AddStringProperty()`属性字符串在设备端和服务器端有没有长度限制，没有MCP接口目前想播放指定的本地歌曲貌似只能将歌曲列表喂到属性里。

还一个问题，也是以后可能会出现的需求，用户说“播放音乐”时服务器端理解后直接就开始播放歌曲了，如果以后上接口用户想通过IOT接口调用自己的音乐API播放或者用IOT能力播放本地音乐，就无法使用最自然的“播放音乐”"我想听xxx"这样的语音控制了。是否可以在小智控制台或者硬件端设备信息的某个特殊字段加个屏蔽服务器音乐的选项，这样识别到播放音乐意图直接调用iot或返回特定格式的数据给硬件自行处理。我目前是通过“播放本地歌曲xxx”来调用iot播放，同时避开服务器音乐的。

如果没有三方接口大佬的服务器音乐库建立估计比较困难，这样服务器音乐能力基本就用不了。设计下接口把音乐能力做起来了小智的实用性应该能提升不少，AI对话目前还是炫技大于实际意义，能控制设备和播放音乐吃灰的情况会少一些，对小智项目发展可能也有益处。

不过不管怎样，还是感谢大佬，前几年接触智能家居硬件的时候用的固定词语音识别芯片控制设备，感觉小爱音响就是天花板了。现在居然能手搓智能音响语义能力还比小爱强得多。

closed 2025-03-26 by A29415

78, 2025-03-26:
> 大佬，MCP大概多久会上？现在让小智控制本地设备和播放本地歌曲都是在IOT属性里喂列表，用 `QwenMax2.5` 它能理解列表提取控制需要的参数，但目前这种方式已经到极限了，QW2.5 理解是最准的，成功率也只能95左右。
>
> 顺便问下 `AddStringProperty()`属性字符串在设备端和服务器端有没有长度限制，没有MCP接口目前想播放指定的本地歌曲貌似只能将歌曲列表喂到属性里。 [...]

已经开始研发，正在加急～～