78/xiaozhi-esp32 #371: 面包板wifi-LCD最新1.4.7版汉字显示bug，顶部汉字无法正常显示

manicwy, 2025-03-19:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.4.7

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

External 5V

### What is the expected behavior?

全部文字能正常显示

### What is the actual behavior?

顶部状态栏汉字显示异常

[image]
[image]

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

isamu2025, 2025-03-19:
表情包改32像素的大小。或者不要开微信对话风格。

manicwy, 2025-03-19:
> 表情包改32像素的大小。或者不要开微信对话风格。

配置都正确的，什么都没改，重新编译了一版，然后重刷固件，就全部正常了，奇怪