78/xiaozhi-esp32 #376: 面包板版本sRGB端口配置错误，

kazu12, 2025-03-20:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.4.7-bread-compact-wifi(-lcd)

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

乐鑫官方开发板ESP32-S3-DevKitC新版本已经调整了sRGB输出管教，老版本sRGB是接到IO48，新的已经修改为IO38，网购很多开发板都是接的IO38，需要修改；

### What is the actual behavior?

如果使用无PSRAM的ESP32-S3模块，在idf.py menuconfig中设置关闭PSRAM后，还是会出现在配置完麦克风和音频解码后不断重启现象，初步判断是某ISR向量指向了外置RAM中，请考虑是否能配置不使用PSRAM的版本，关闭PSRAM后能调整指向到内置RAM中

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

78, 2025-03-20:
IO48改到IO38应该是一个很重大的更新了，硬件发生了变化，不能简单的通过修改宏配置来适配。

kazu12, 2025-03-21:
是否可以单开一个版本，正对不同版本dev模块设置？

kazu12, 2025-03-28:
[官方说明请见地址：](https://docs.espressif.com/projects/esp-dev-kits/zh_CN/latest/esp32s3/esp32-s3-devkitc-1/user_guide.html#id19)
 
`目前首次发布和 v1.1 版本的开发板均可订购，两个版本的主要差异在于 RGB LED 所连接的管脚不同。首次发布版本的 RGB LED 连接到 GPIO48，而 v1.1 版本的 RGB LED 连接到 GPIO38。`

78, 2025-03-28:
> [官方说明请见地址：](https://docs.espressif.com/projects/esp-dev-kits/zh_CN/latest/esp32s3/esp32-s3-devkitc-1/user_guide.html#id19)
> 
> `目前首次发布和 v1.1 版本的开发板均可订购，两个版本的主要差异在于 RGB LED 所连接的管脚不同。首次发布版本的 RGB LED 连接到 GPIO48，而 v1.1 版本的 RGB LED 连接到 GPIO38。`

可以考虑在 config.json 里创建一个 build，生成一个新的固件，通过配置一个宏来解决这个兼容性。