78/xiaozhi-esp32 #377: 执行IOT指令时莫名会收到goodbye，回到待命

dopheee, 2025-03-20:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

仅尝试V1.1.2和V1.4.7，均存在

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

正常执行IOT指令，如背光调节、音量调节

### What is the actual behavior?

[image]
啥也没执行，直接收到goodbye回到待命

### Steps to reproduce.

反复调节音量或背光即可复现

### Debug Logs.

```plain

```

78, 2025-03-20:
看起来像是模型罢工了，输出了退出的指令。要看一下聊天记录。

dopheee, 2025-03-20:
> 看起来像是模型罢工了，输出了退出的指令。要看一下聊天记录。

[image]

[image]
我觉得挺容易触发的，多控制几次就触发了

isamu2025, 2025-03-20:
因该是deeepseekV3

dopheee, 2025-03-20:
> 确实是模型问题，你选择的是哪一个？

DeepSeek V3

78, 2025-03-20:
> 因该是deeepseekV3

同上，V3有JSON格式错误的问题，解析错误导致服务器报错，然后断开了连接

78, 2025-03-21:
对DeepSeek V3打了补丁，可以看看这个问题是否还存在