78/xiaozhi-esp32 #378: 使用pcm5102输出音频，只有左边声道有声音

uduoy, 2025-03-20:
插的3.5mm耳机。相关代码尝试改过，但是音质会失真，请教应该如何处理？针脚用的原max98357针脚

uduoy, 2025-03-20:
打通左？是否就把左复制到右也行？具体应该如何做呢

sevenberyl, 2025-03-20:
可以自己实现一个_audio_codec类，重写Write方法

sevenberyl, 2025-03-21:
> > 可以自己实现一个_audio_codec类，重写Write方法
> 
> 你会吗？我几乎不会编程，现在只能让cursor改😄

大概是这样，你自己调试一下：
size_t bytes_written;
    // Convert mono to stereo by duplicating the samples
    std::vector<int32_t> stereo_buffer(samples * 2);
    for (int i = 0; i < samples; i++) {
        stereo_buffer[2 * i] = buffer[i];
        stereo_buffer[2 * i + 1] = buffer[i];
    }

    ESP_ERROR_CHECK(i2s_channel_write(tx_handle_, stereo_buffer.data(), stereo_buffer.size() * sizeof(int32_t), &bytes_written, portMAX_DELAY));
    return bytes_written / sizeof(int32_t);

jovelove, 2025-09-10:
希望支持双声道！就可以有立体声了。虾哥威武！