78/xiaozhi-esp32 #386: XMINI-C3结合asrpro的设计改进

winslow49, 2025-03-21:
### Is your feature request related to a problem?

目前可以用asrpro开发板唤醒XMINI-C3，其实很简单，就是把asrpro的中断信号接到了XMINI-C3的按键上，但XMINI-C3收到中断后还要继续说话才能答复，有没有办法让XMINI-C3收到中断信号后就直接答复呢？

78, 2025-03-21:
> ### Is your feature request related to a problem?
> 目前可以用asrpro开发板唤醒XMINI-C3，其实很简单，就是把asrpro的中断信号接到了XMINI-C3的按键上，但XMINI-C3收到中断后还要继续说话才能答复，有没有办法让XMINI-C3收到中断信号后就直接答复呢？
>
> ### Describe the solution you'd [...]

改一行代码就可以，要自己实现，参考教程里的相关文档吧

huangrisheng, 2025-03-25:
能指出来是那句吗？没有找到