78/xiaozhi-esp32 #388: fatal error: assets/lang_config.h: No such file or directory

ludashi6789, 2025-03-21:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

lichuang-dev-1.4.6

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

 fatal error: assets/lang_config.h: No such file or directory
编译出现问题，缺少这个文件  版本是\xiaozhi-esp32-1.4.6

[image]

### What is the actual behavior?

 fatal error: assets/lang_config.h: No such file or directory
编译出现问题，缺少这个文件  版本是\xiaozhi-esp32-1.4.6

[image]

### Steps to reproduce.

 fatal error: assets/lang_config.h: No such file or directory
编译出现问题，缺少这个文件  版本是\xiaozhi-esp32-1.4.6

[image]

### Debug Logs.

```plain

```

KingingWang, 2025-03-25:
1.5.2 same

KingingWang, 2025-03-25:
我刚刚看了一下,你可能和我使用了默认的esp32,请使用`idf.py set-target esp32s3` 修改目标,记得清理工程

ludashi6789, 2025-03-25:
> 我刚刚看了一下,你可能和我使用了默认的esp32,请使用`idf.py set-target esp32s3` 修改目标,记得清理工程

不是这个问题，我修改了这个，依旧报错，另外指向的是assets/lang_config.h 项目目录下没有这个

78, 2025-03-26:
> > 我刚刚看了一下,你可能和我使用了默认的esp32,请使用`idf.py set-target esp32s3` 修改目标,记得清理工程
> 
> 不是这个问题，我修改了这个，依旧报错，另外指向的是assets/lang_config.h 项目目录下没有这个

这个是build的时候会自动生成。你可以在项目中搜索 lang_config.h

closed 2025-04-09 by ludashi6789

ludashi6789, 2025-04-10:
感谢 解决方法如下
1、先进入menuconfig 选中下语音  save 保存
2、先执行build方法 确认项目目录下生成了lang_config.h
3、然后执行idf.py flash monitor即可

[image]