78/xiaozhi-esp32 #389: 新面包板支持金逸晨1.54寸TFT液晶屏ST7789P3小屏幕240*240模块LCD串口SPI彩屏显示

dreamchen, 2025-03-21:
### Is your feature request related to a problem?

[image]
该显示屏按面包板WIFI-LCD彩屏接线教程接线后仅背光亮，无法显示任何内容

sakeeeeeeeee, 2025-03-23:
我也遇到了

sevenberyl, 2025-03-25:
我的1.41版本没有问题，配置如下：

```
#define` DISPLAY_OFFSET_X 0
#define DISPLAY_OFFSET_Y 0
#define DISPLAY_WIDTH    240
#define DISPLAY_HEIGHT   240
#define DISPLAY_SWAP_XY  false
#define DISPLAY_MIRROR_X false
#define DISPLAY_MIRROR_Y false

#define DISPLAY_BACKLIGHT_PIN GPIO_NUM_5//GPIO_NUM_NC
#define DISPLAY_BACKLIGHT_OUTPUT_INVERT true
```

```
void Initialize_ST7789_240X240_Display()
    {
        ESP_LOGI(TAG, "Initializing lcd display");
        spi_host_device_t SPI_HOST = SPI2_HOST;
        ESP_LOGI(TAG, "Initializing lcd spi bus");
        spi_bus_config_t buscfg = {};
        buscfg.mosi_io_num = LCD_NUM_MOSI;
        buscfg.miso_io_num = GPIO_NUM_NC;
        buscfg.sclk_io_num = LCD_NUM_SCLK;
        buscfg.quadwp_io_num = GPIO_NUM_NC;
        buscfg.quadhd_io_num = GPIO_NUM_NC;
        buscfg.max_transfer_sz = DISPLAY_WIDTH * DISPLAY_HEIGHT * sizeof(uint16_t);
        ESP_ERROR_CHECK(spi_bus_initialize(SPI_HOST, &buscfg, SPI_DMA_CH_AUTO));

        ESP_LOGD(TAG, "Install panel IO");
        esp_lcd_panel_io_handle_t panel_io = nullptr;
        esp_lcd_panel_handle_t panel = nullptr;
        esp_lcd_panel_io_spi_config_t io_config = {};
        io_config.cs_gpio_num = LCD_NUM_CS;
        io_config.dc_gpio_num = LCD_NUM_DC;
        io_config.spi_mode = 2;
        io_config.pclk_hz = 20 * 1000 * 1000;
        io_config.trans_queue_depth = 10;
        io_config.lcd_cmd_bits = 8;
        io_config.lcd_param_bits = 8;
        ESP_ERROR_CHECK(esp_lcd_new_panel_io_spi(SPI_HOST, &io_config, &panel_io));

        ESP_LOGD(TAG, "Install LCD driver");
        esp_lcd_panel_dev_config_t panel_config = {};
        panel_config.reset_gpio_num = LCD_NUM_RST;
        panel_config.rgb_ele_order = LCD_RGB_ELEMENT_ORDER_RGB;
        panel_config.bits_per_pixel = 16;
        ESP_ERROR_CHECK(esp_lcd_new_panel_st7789(panel_io, &panel_config, &panel));
        ESP_ERROR_CHECK(esp_lcd_panel_reset(panel));
        ESP_ERROR_CHECK(esp_lcd_panel_init(panel));
        ESP_ERROR_CHECK(esp_lcd_panel_swap_xy(panel, DISPLAY_SWAP_XY));
        ESP_ERROR_CHECK(esp_lcd_panel_mirror(panel, DISPLAY_MIRROR_X, DISPLAY_MIRROR_Y));
        ESP_ERROR_CHECK(esp_lcd_panel_invert_color(panel, true));

        display_ = new SpiLcdDisplay(panel_io, panel,
                                     DISPLAY_BACKLIGHT_PIN,
                                     DISPLAY_BACKLIGHT_OUTPUT_INVERT,
                                     DISPLAY_WIDTH,
                                     DISPLAY_HEIGHT,
                                     DISPLAY_OFFSET_X,
                                     DISPLAY_OFFSET_Y,
                                     DISPLAY_MIRROR_X,
                                     DISPLAY_MIRROR_Y,
                                     DISPLAY_SWAP_XY,
                                     {
                                         .text_font = &font_puhui_20_4,
                                         .icon_font = &font_awesome_20_4,
                                         .emoji_font = font_emoji_64_init(),
                                     });
    }

```

dreamchen, 2025-03-27:
跳线问题，已经OK。感谢。

closed 2025-03-27 by dreamchen

zunlongzhou, 2025-05-11:
> 跳线问题，已经OK。感谢。

请问怎么解决的呢，我的板子是esp32-s3，也是8pin屏幕，但是背光亮屏幕不亮，应该怎么接线呢，我看到吗cs接22口，但是板子上没有22口。接线方便告知一下吗