78/xiaozhi-esp32 #393: 蓝牙音频功能

vic-gz, 2025-03-22:
### Is your feature request related to a problem?

有没有计划加入蓝牙音频的功能？
例如在支持通话功能的音箱上，用蓝牙录音和播放替代现在的I2S进行录音和播放。
在汽车上甚至可以通过蓝牙歌词相关技术来控制车机屏幕显示对话内容

9cat, 2025-03-24:
非常需要这个。 可能 而且不用线板上的  音箱与MIC。 直接用蓝牙作为输入输出。

diaoling6665, 2025-03-26:
同求同求

tangyumei3535, 2025-04-02:
经典蓝牙才可以支持HFP或A2DP协议哦

mcdullmedia, 2025-04-07:
嗯. 我的 esp32 沒有麥克也沒有擴音.

ForgetCSX, 2025-04-07:
这样就有交互的问题，现在大多数板子都没有这种交互界面，如何去选择蓝牙设备，做成和配网的界面一样的嘛？

sunnythree, 2025-04-13:
同求

xunnew18, 2025-06-23:
同求

pangyubo, 2026-03-30:
非常需求   只有要蓝牙发射就可以

rocaltair, 2026-04-02:
@78 @cqrect @ForgetCSX @tangyumei3535 

我有一个新的思路，现在几乎都是 type c 接口，如果直接在 ROM 里实现 type c 蓝牙适配器的驱动功能，这样所有的 esp32 设备可以直接通过插适配器来实现耳麦的交互？

通常来说，一个这样的适配器也就几块钱，并且可用性大大增加，所有现有的设备升级下 ROM 就能无缝用起来。对于终端用户动手能力再弱，插个适配器总不至于不会吧。