78/xiaozhi-esp32 #394: Kconfig.projbuild配置冲突

GreedyPirate, 2025-03-23:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.5.0-1-g11c79bf

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

idf.py menuconfig选择完board和屏幕后应该生效

### What is the actual behavior?

按s保存之后执行idf.py build，出现冲突，导致sdkconfig文件被重命名sdkconfig.old(vscode也可以看到过程), BOARD_TYPE被重置为default值，屏幕类型也被重置为OLED_SSD1306_128X32

### Steps to reproduce.

BOARD_TYPE选择BOARD_TYPE_BREAD_COMPACT_WIFI_LCD之后，DISPLAY_LCD_TYPE选择LCD_GC9A01_240X240

### Build or installation Logs.

```plain

```

### More Information.

原因分析
DISPLAY_OLED_TYPE没有关联BOARD_TYPE_BREAD_COMPACT_WIFI_LCD，默认设为了OLED_SSD1306_128X32，导致和LCD_GC9A01_240X240冲突，idf.build 检测到冲突重新生成了整个sdkconfig，最终导致LCD_GC9A01_240X240配置丢失

我的临时解决办法，增加!BOARD_TYPE_BREAD_COMPACT_WIFI_LCD判断，但只是这一种情况，其他的希望也修复下
`
choice DISPLAY_OLED_TYPE
    depends on (BOARD_TYPE_BREAD_COMPACT_WIFI || BOARD_TYPE_BREAD_COMPACT_ML307 || BOARD_TYPE_BREAD_COMPACT_ESP32) && !BOARD_TYPE_BREAD_COMPACT_WIFI_LCD
    prompt "OLED Type"
    default OLED_SSD1306_128X32
    help
        OLED 屏幕类型选择
    config OLED_SSD1306_128X32
        bool "SSD1306, 分辨率128*32"
    config OLED_SSD1306_128X64
        bool "SSD1306, 分辨率128*64"
endchoice
`

78, 2025-03-23:
@ooxxU 看看是bug吗

ooxxU, 2025-03-24:
> [@ooxxU](https://github.com/ooxxU) 看看是bug吗

按照描述的步骤操作了，并没有出现冲突：
用的是1.5.2的最新代码，
板子选择的是S3的面包板WIFI-LCD
分辨率选择的是 LCD_GC9A01_240X240，点击 SAVE 没有报错和冲突
如下图：
[image]

ooxxU, 2025-03-24:
> [@ooxxU](https://github.com/ooxxU) 看看是bug吗

build了下，也是成功的

[image]

closed 2025-03-29 by GreedyPirate