78/xiaozhi-esp32 #400: 繁体中文显示方框

neil0306, 2025-03-24:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.5.2_esp-box-3.zip

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Power Supply used.

Battery

### What is the expected behavior?

希望可以正确显示繁体字。

### What is the actual behavior?

目前我使用的硬件是乐鑫的box3, 我编译固件的时候用idf.py menuconfig 选择了繁体中文（zh-TW），刷入之后我发现本应该显示繁体字的地方用方框来替代了， 想知道有没有什么解决办法吗？ 

[image]

### Steps to reproduce.

按照官方教程，选择硬件型号为乐鑫的box3，选择zh-TW作为字体，然后按照教程编译，用浏览器刷入固件。

### Debug Logs.

```plain

```

xuelse, 2025-03-24:
字库缺失这个字，所以就是框，要完善字库

78, 2025-03-24:
目前只能在 xiaozhi-fonts 里生成新的字体文件来解决，需要提交PR到那个project。

neil0306, 2025-03-25:
> 目前只能在 xiaozhi-fonts 里生成新的字体文件来解决，需要提交PR到那个project。

好的，我尝试研究一下怎么解决~

neil0306, 2025-03-25:
@78 虾哥你好，我已经生成了对应的`.c`字体文件，并提交了PR，但由于xiaozhi-fonts 中没有提供生成`.p3`文件的相关脚本，我无法直接编译固件来测试实际效果，麻烦您查看一下是否可用，如果不可用麻烦您提供一下反馈，谢谢~

ReMakeDoG, 2025-03-30:
請問最後有找到方法嗎?
@78 @neil0306 
我有嘗試將
1. SourceHanSansTC-VF.ttf字體包
2. 加入Big5.txt 轉換成unicode
3. 修改 manged_components 內的 font.py、generate_font.py、CMakeList.txt
4. 修改 main 內的 boards、display

78, 2025-03-30:
理论上只需要扩充GB2312.txt的内容，增加繁体字即可。

ReMakeDoG, 2025-03-30:
@78 你好，我有嘗試過了，也不行還是一樣，請問會有教學影片嗎?或是比較詳細的做法?

lewis50093, 2025-04-07:
@78 蝦哥您好 我也遇到了繁體方框的問題 想問有沒有更詳細的解決方式呢？ 謝謝

78, 2025-04-07:
> [@78](https://github.com/78) 蝦哥您好 我也遇到了繁體方框的問題 想問有沒有更詳細的解決方式呢？ 謝謝

目前群友的做法是重新生成繁体字体，2.0的版本考虑做成云端下载字体，不然固件太大了，无法OTA了。

lewis50093, 2025-04-08:
> > [@78](https://github.com/78) 蝦哥您好 我也遇到了繁體方框的問題 想問有沒有更詳細的解決方式呢？ 謝謝
> 
> 目前群友的做法是重新生成繁体字体，2.0的版本考虑做成云端下载字体，不然固件太大了，无法OTA了。

謝謝蝦哥 有重新生成繁體字體，已可正常顯示 謝謝

neil0306, 2025-04-08:
> > > [@78](https://github.com/78) 蝦哥您好 我也遇到了繁體方框的問題 想問有沒有更詳細的解決方式呢？ 謝謝
> > 
> > 
> > 目前群友的做法是重新生成繁体字体，2.0的版本考虑做成云端下载字体，不然固件太大了，无法OTA了。
> 
> 謝謝蝦哥 有重新生成繁體字體，已可正常顯示 謝謝

我之前尝试了添加繁体字（如big5编码和香港增补字库），这时候只有繁体字是可以正常显示的，但此时的另一个问题是偶尔LLM的回答是简繁混合的，这就导致简体字会莫名变成另一个字，比如简体字“吗”变成其他字体。

我目前用的是自己部署的后端服务器，这时候我可以用opencc这类库强行把所有字体转成繁体……

如果用虾哥的服务器，我目前就只能把简繁体都添加到一起，最终就如虾哥所说，固件太大会刷机失败……

lewis50093, 2025-04-09:
> 我之前尝试了添加繁体字（如big5编码和香港增补字库），这时候只有繁体字是可以正常显示的，但此时的另一个问题是偶尔LLM的回答是简繁混合的，这就导致简体字会莫名变成另一个字，比如简体字“吗”变成其他字体。
>
> 我目前用的是自己部署的后端服务器，这时候我可以用opencc这类库强行把所有字体转成繁体……
>
> 如果用虾哥的服务器，我目前就只能把简繁体都添加到一起，最终就如虾哥所说，固件太大会刷机失败……

沒錯 我也是用跟您一樣的方法 STT TTS都先翻譯成繁體 可以正常 但我請他講日文韓文就會變亂碼🤣🤣