78/xiaozhi-esp32 #403: 1.5.2 版本编译报错 class Display' has no member named 'brightness'

sevenberyl, 2025-03-25:
1.5.2 编译时，display缺少方法，编译报错

D:/personal/Arduino/06-xiaozhi/esp32_ai/main/iot/things/blaklight.cc
D:/personal/Arduino/06-xiaozhi/esp32_ai/main/iot/things/blaklight.cc: In lambda function:
D:/personal/Arduino/06-xiaozhi/esp32_ai/main/iot/things/blaklight.cc:19:29: error: 'class Display' has no member named 'brightness'
   19 |             return display->brightness();
      |                             ^~~~~~~~~~
D:/personal/Arduino/06-xiaozhi/esp32_ai/main/iot/things/blaklight.cc: In lambda function:
D:/personal/Arduino/06-xiaozhi/esp32_ai/main/iot/things/blaklight.cc:28:22: error: 'class Display' has no member named 'SetBacklight'
   28 |             display->SetBacklight(brightness);
      |                      ^~~~~~~~~~~~
[1794/1826] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/boards/common/wifi_board.cc.obj
ninja: build stopped: subcommand failed.

sevenberyl, 2025-03-25:
新版本是需要自己实现display派生类了吗？？
看不到dispaly类的 SetBacklight(brightness); 和 display->brightness(); 方法呢

78, 2025-03-25:
看一下其他板子的实现方法

sevenberyl, 2025-03-26:
> 看一下其他板子的实现方法

请问可以参考哪个板子？搜了一下没发现有引用的

78, 2025-03-26:
> > 看一下其他板子的实现方法
> 
> 请问可以参考哪个板子？搜了一下没发现有引用的

搜 Backlight 肯定有