78/xiaozhi-esp32 #413: Button类DoubleClick回调函数不执行

ACI0419, 2025-03-26:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.4.6 wifi-compact-board

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Power Supply used.

USB

### What is the expected behavior?

button类(使用volume button down测试)注册DoubleClick回调函数，双击button时应该会调用回调函数

### What is the actual behavior?

双击按键时，不管怎么按都识别为两次single click，不执行double click回调函数

### Steps to reproduce.

1. 原有代码基础上，volume button up注册DoubleClick回调函数
2. 烧录并重启
3. 双击button
...

### Debug Logs.

```plain

```

### More Information.

1.尝试用最新的（4.1.1）IoT Button,官方demo DoubleClick正常，但是4.1.1版本和现有的3.5.0不兼容，没有在小智上进一步测试
2.使用目前小智用的3.5.0版本IoT Button,IoT Button官方demo工作正常，小智上不正常，出现如上错误

后面找到问题了，实测把button.cc中的button_config中的press_time注释掉，问题解决
button_config_t button_config = {
        .type = BUTTON_TYPE_GPIO,
        // .long_press_time = 1000, //使用这两个参数会导致双击失效
        // .short_press_time = 50,
        .gpio_button_config = {
            .gpio_num = gpio_num,
            .active_level = static_cast<uint8_t>(active_high ? 1 : 0)
        }
    };

78, 2025-03-26:
这两个的默认参数是多少吗？

isamu2025, 2025-03-26:
用小智云聊的代码

void Button::OnDoubleClick(std::function<void()> callback) {
    if (button_handle_ == nullptr) {
        return;
    }
    on_double_click_ = callback;
    iot_button_register_cb(button_handle_, BUTTON_DOUBLE_CLICK, [](void* handle, void* usr_data) {
        Button* button = static_cast<Button*>(usr_data);
        if (button->on_double_click_) {
            button->on_double_click_();
        }
    }, this);
}

void XiaoZhiYunliaoC3::InitializeButtons() {

boot_button_.OnDoubleClick([this]() {
        // ESP_LOGI(TAG, "Button OnDoubleClick");
        if (display_ && !wifi_config_mode_) {
            display_->SwitchPage();
        }
    });  

｝

ACI0419, 2025-04-03:
> 这两个的默认参数是多少吗？
在espresssif__button/include/iot_button.h里面定义的：

typedef enum {
    BUTTON_LONG_PRESS_TIME_MS = 0,
    BUTTON_SHORT_PRESS_TIME_MS,
    BUTTON_PARAM_MAX,
} button_param_t;

typedef struct {
    button_type_t type;                               /**< button type, The corresponding button configuration must be filled */
    uint16_t long_press_time;                         /**< Trigger time(ms) for long press, if 0 default to BUTTON_LONG_PRESS_TIME_MS */
    uint16_t short_press_time;                        /**< Trigger time(ms) for short press, if 0 default to BUTTON_SHORT_PRESS_TIME_MS */
    union {
        button_gpio_config_t gpio_button_config;      /**< gpio button configuration */
#if CONFIG_SOC_ADC_SUPPORTED
        button_adc_config_t adc_button_config;        /**< adc button configuration */
#endif
        button_matrix_config_t matrix_button_config; /**< matrix key button configuration */
        button_custom_config_t custom_button_config;  /**< custom button configuration */
    }; /**< button configuration */
} button_config_t;

ACI0419, 2025-04-03:
> 用小智云聊的代码
>
> void Button::OnDoubleClick(std::function<void()> callback) { if (button_handle_ == nullptr) { return; } on_double_click_ = callback; iot_button_register_cb(button_handle_, [...]

你这段代码是注册和执行回调函数的，但是虾哥这个的回调函数没问题，不执行是因为iot button不识别DoubleClick事件

A29415, 2025-04-09:
在按键初始化时不对short_press_time赋值，让它使用默认值，我这边测试可以触发DoubleClick

[image]

78, 2025-04-09:
> 在按键初始化时不对short_press_time赋值，让它使用默认值，我这边测试可以触发DoubleClick
> 
> [image]

不设置的话，它默认是什么值？

A29415, 2025-04-09:
> > 在按键初始化时不对short_press_time赋值，让它使用默认值，我这边测试可以触发DoubleClick
> > [image]
> 
> 不设置的话，它默认是什么值？

应该是这个值

78, 2025-04-09:
`1` 么

A29415, 2025-04-09:
> `1` 么
对是1，但是有个奇怪的点，不给它赋值可以检测到双击，给它赋值1或者BUTTON_SHORT_PRESS_TIME_MS，就检测不到双击

78, 2025-04-09:
要看代码实现了，都是有代码的

78, 2025-05-20:
新版本已解决。

closed 2025-05-20 by 78