78/xiaozhi-esp32 #421: 神奇按钮2.4运行1.5.2固件。暗黑模式返回正常模式后显示异常

zhu0312, 2025-03-28:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

神奇按钮2.4，固件1.5.2

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Power Supply used.

USB

### What is the expected behavior?

神奇按钮2.4运行1.5.2固件。暗黑模式返回正常模式后显示异常

### What is the actual behavior?

神奇按钮2.4运行1.5.2固件。暗黑模式返回正常模式后显示异常

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

zhu0312, 2025-04-02:
因为这个固件本来就是黑底的，语音使它切换暗黑模式后。显示就偏白，语音使它返回正常模式后显示是黑白混在一起的样子，附件是图片。如果和传统模式一样的图片就好了，还有就是可不可以显示电量。

zhu0312, 2025-04-02:
[image]
[image]
[image]

78, 2025-04-02:
这个可能跟2.4的屏幕有关，2.5版本修复了这个问题，硬件也发生了变化。

closed 2025-05-20 by 78