78/xiaozhi-esp32 #426: 编译后启动异常

G-Akiraka, 2025-03-29:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.5.2

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

正常启动, 通过下载的固件 v1.5.2_movecall-moji-esp32s3.zip 可以正常启动，自己编译则不行，代码没做其他调整

### What is the actual behavior?

疯狂报 0xffffffff

### Steps to reproduce.

开发板 esp32s3 N16R8
1、idf.py set-target esp32s3
2、idf.py menuconfig 设置需要的板子(moji)，和flash 改为16m
3、执行 idf.py build
4、cd scripts && python release.py
5、烧录 build 下的 merged-binary.bin

### Build or installation Logs.

```plain
![Image](https://github.com/user-attachments/assets/fcbed726-ed78-4591-979f-e8d4c91fa182)
```

78, 2025-03-29:
release.py 后面加上板子类型

G-Akiraka, 2025-03-30:
> release.py 后面加上板子类型

可以了 感谢

closed 2025-03-30 by G-Akiraka