78/xiaozhi-esp32 #435: 能否支持Lottie动画显示表情，或者PNG序列图片，资源数据可以存储在SD卡

fengjiannan2010, 2025-04-01:
### Is your feature request related to a problem?

能否支持Lottie动画显示表情，或者PNG序列图片，资源数据可以存储在SD卡
类似MagAI硬件一样，可以自定义多种动画或播放PNG图片序列。
【Magai适配最新版本了～】 https://www.bilibili.com/video/BV1PVXuY6ERr/?share_source=copy_web&vd_source=60d5dd47b0e94d365ae6bb507ab25a44

78, 2025-04-01:
lottie动画的性能还有待测试，担心动画播放时，会影响音频编解码。

zyl2995230, 2025-04-07:
我采用gif格式进行表情播放没有出现音频解码问题，一切正常！

qaz624824554, 2025-04-11:
gif我试过可以，但是lottie只要一播放音频就崩溃，目前还没找到解决方案

78, 2025-04-11:
@ZhouKes 播放lottie会影响到音频吗？

qaz624824554, 2025-04-13:
lvgl9.2使用的是ThorVG作为lottie渲染器，而ThorVG占用的堆栈很大，调试发现占用保持在 20000 word左右，相当于 20000 * 4 = 80000 byte以上，而 gif 渲染器才 1000 多 word，因此 ThorVG 要分配 4096 * 25 = 102400 byte = 1024 KB = 1MB 的堆栈任务才比较保险，但不知道为啥，一直分配失败，明明我的 PSRAM 有 8MB，有大佬知道有哪些解决方案吗

ZhouKes, 2025-04-13:
@78 lottie占用的任务栈很大，确实可能影响其它任务的执行。
@qaz624824554 是不是任务栈不能使用PSRAM？

78, 2025-04-13:
> lvgl9.2使用的是ThorVG作为lottie渲染器，而ThorVG占用的堆栈很大，调试发现占用保持在 20000 word左右，相当于 20000 * 4 = 80000 byte以上，而 gif 渲染器才 1000 多 word，因此 ThorVG 要分配 4096 * 25 = 102400 byte = 1024 KB = 1MB 的堆栈任务才比较保险，但不知道为啥，一直分配失败，明明我的 PSRAM 有 8MB，有大佬知道有哪些解决方案吗

用 heap_caps_malloc 分配 PSRAM 的1MB内存，然后用 xTaskCreateStatic 分配 task 呢？

qaz624824554, 2025-04-14:
> > lvgl9.2使用的是ThorVG作为lottie渲染器，而ThorVG占用的堆栈很大，调试发现占用保持在 20000 word左右，相当于 20000 * 4 = 80000 byte以上，而 gif 渲染器才 1000 多 word，因此 ThorVG 要分配 4096 * 25 = 102400 byte = 1024 KB = 1MB 的堆栈任务才比较保险，但不知道为啥，一直分配失败，明明我的 PSRAM 有 8MB，有大佬知道有哪些解决方案吗
> 
> 用 heap_caps_malloc 分配 PSRAM 的1MB内存，然后用 xTaskCreateStatic 分配 task 呢？

试过了，也不行，heap_caps_malloc 可以分配成功，但是xTaskCreateStatic 不起作用

78, 2025-04-14:
应该是堆栈不能大于64KB，可以问一下官方的人。

qaz624824554, 2025-04-14:
那估计用不了，我上面计算错了，是100KB的堆栈

78, 2025-05-13:
会是 lvgl 的 task stack size 不足么

dykwok, 2025-05-14:
> 会是 lvgl 的 task stack size 不足么

参考上面我分享的工程，我把 `ESP_TIMER_TASK_STACK_SIZE` 和 `ESP_MAIN_TASK_STACK_SIZE` 这两个参数都设置到 `40960` 可以播 `120x120` 了，再往上也不行了。

junhuanchen, 2025-07-22:
刚整完，给后来人一些参考，最新的想要测试动画效果参考，测试尺寸 360*360 

commit eb0bba2c89c00503667d8e41dad5a94e15191fce (HEAD -> main, origin/main, origin/HEAD)
Author: Xiaoxia <terrence@tenclass.com>
Date:   Tue Jul 22 10:06:11 2025 +0800

    Update LICENSE

硬件

WaveshareEsp32s3TouchAMOLED1inch8

效果

https://github.com/user-attachments/assets/cccd055e-a573-4453-8433-4ce1a2dbb741

日志

[image]

qaz624824554, 2025-07-23:
> 刚整完，给后来人一些参考，最新的想要测试动画效果参考，测试尺寸 360*360
> 
> commit [eb0bba2](https://github.com/78/xiaozhi-esp32/commit/eb0bba2c89c00503667d8e41dad5a94e15191fce) (HEAD -> main, origin/main, origin/HEAD) Author: Xiaoxia [terrence@tenclass.com](mailto:terrence@tenclass.com) Date: Tue Jul 22 10:06:11 2025 +0800
> 
> ```
> Update LICENSE
> ```
> 
> 硬件
> 
> WaveshareEsp32s3TouchAMOLED1inch8
> 
> 效果
> 
>  9067832a472705e25e7d6798e6eb0eea.mp4 
> 日志
> 
> <img alt="Image" width="1012" height="899" src="https://private-user-images.githubusercontent.com/32978053/469023020-6bee3c10-72b8-4a7f-b687-2610cba30f52.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTMyNjI0MDUsIm5iZiI6MTc1MzI2MjEwNSwicGF0aCI6Ii8zMjk3ODA1My80NjkwMjMwMjAtNmJlZTNjMTAtNzJiOC00YTdmLWI2ODctMjYxMGNiYTMwZjUyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzIzVDA5MTUwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRkZmMwNmQ5OWU0YmY2ZDBlZGNmYjcxN2U1MDVkNzI5NDUwMzQzYWRjNmE2NzYzOWNmZjJiOTExNGI1YzMyNWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.TPJyoADtAr1-zkuFO48hCBcZvSCrtB5410sVwIM6Nfw">

请问有出现程序崩溃情况吗，我之前也可以正常显示，但是会有偶发崩溃的现象

junhuanchen, 2025-08-13:
我的最新效果供参考

https://github.com/user-attachments/assets/75b0a091-ee6f-4f34-867e-372fc572c30e

zhiyoutech, 2025-09-08:
请参考：https://blog.gitcode.com/db8e14e0e806d4e76a9ddcce3ddcf4a7.html