78/xiaozhi-esp32 #437: 希望支持训练ESP-32S3的自定义唤醒词

chapterv, 2025-04-01:
### Is your feature request related to a problem?

目前看到调整唤醒词的方法有：
1. 小智百科里面的使用免费26个唤醒词之一 https://rcnv1t9vps13.feishu.cn/wiki/H4dywK6ucigEtykKmm3cmZ7Dnog
2. 使用额外的asrpro板子加外挂

以上方案都不太理想，要么没法自定义，要么是要额外配置板子。

### Describe the solution you'd like.

然后看到这个贴子给出了**完整的ESP32-S3 N16R8的唤醒词训练过程，能否考虑对接整合到小智AI中**？
fork项目git地址：https://gitee.com/zhuxiaohuaqn/esp32s3-ai-chat
fork项目文字教程：https://www.qingningz.cn/archives/2462
原版项目视频（14-16）：
https://www.bilibili.com/video/BV1UXx7eDEF7
https://www.bilibili.com/video/BV1TXx7eQE6h
https://www.bilibili.com/video/BV1mXx7eDEdb

78, 2025-04-01:
确实可以做一个这方面的方案，整合到现在的教程里

chapterv, 2025-04-02:
多谢虾哥回复，支持大佬！！

78, 2025-04-02:
> 多谢虾哥回复，支持大佬！！

你也可以先实践一下，对比其他方案，看看效果怎么样

Kevincoooool, 2025-04-03:
试过了效果不好

GGBond-next, 2025-05-12:
在网站上训练好了自己的模型之后应该怎么实现小智唤醒词的更换呢

78, 2025-05-12:
> 在网站上训练好了自己的模型之后应该怎么实现小智唤醒词的更换呢

需要自己修改代码实现