78/xiaozhi-esp32 #441: 立创S3的板子，按最右边的打断按键后，屏幕显示待命，这时候需要叫两次小智才能唤醒

Ariken2002, 2025-04-03:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

1.5.5

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Power Supply used.

USB

### What is the expected behavior?

按右边按键打断后，应该可以马上唤醒，而不是需要叫两次。

### What is the actual behavior?

立创S3的板子，按最右边的打断按键后，屏幕显示待命，这时候需要叫两次小智才能唤醒，这是bug么？叫第一次，屏幕会闪一下“连接中”，然后立马变成待命状态，需要再叫一次才有反应。

能稳定复现，待命后，第一次叫小智，会出现“连接中”-“聆听中”，一闪而过，马上又到待命状态了，第二次叫就能正常唤醒了。

### Steps to reproduce.

立创S3的板子，按最右边的打断按键后，屏幕显示待命，这时候需要叫两次小智才能唤醒，这是bug么？叫第一次，屏幕会闪一下“连接中”，然后立马变成待命状态，需要再叫一次才有反应。

能稳定复现，待命后，第一次叫小智，会出现“连接中”-“聆听中”，一闪而过，马上又到待命状态了，第二次叫就能正常唤醒了。

### Debug Logs.

```plain

```

Ariken2002, 2025-04-03:
用的是1.5.5最新版，用的ESP-IDF 5.3 PowerShell编译的，以前的版本好像没有这个问题，也可能没太注意，因为这个打断功能之前用得不多。

78, 2025-04-03:
是存在这个问题，看是否都改成单击多轮对话的模式？ （目前应该是长按说话）

Ariken2002, 2025-04-03:
> 是存在这个问题，看是否都改成单击多轮对话的模式？ （目前应该是长按说话）

单击让小智变成待命这个功能设计我感觉没问题，尤其是没有开启AEC的时候，方便手动打断，只是它目前是单击后，要两次才能唤醒，看上去是个bug。

jinwyp, 2025-04-05:
立创S3 1.5.6 测试 没有这个问题 待命后  一次 小智 就可以唤醒

Ariken2002, 2025-04-05:
> 立创S3 1.5.6 测试 没有这个问题 待命后 一次 小智 就可以唤醒

谢谢提醒，感谢虾哥更新，我试试看。

closed 2025-05-20 by 78