78/xiaozhi-esp32 #448: idf.py menuconfig看不到realtime 选项

Alex-toughman, 2025-04-07:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.5.5-v1.5.6_2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I use idf.py menuconfig to build binary file, I found there is a "USE_REALTIME_CHAT" entry in source code "main/Kconfig.projbuild" but I can't found "USE_REALTIME_CHAT" option in menu window?

### What is the actual behavior?

[image]

### Steps to reproduce.

1. Step idf.py menuconfig

...

### Build or installation Logs.

```plain

```

Alex-toughman, 2025-04-07:
sorry 刚才看到如果borad type选择ESP BoX3那个选项就会出现，请问这是因为面包板的硬件不支持realtime吗

78, 2025-04-07:
> sorry 刚才看到如果borad type选择ESP BoX3那个选项就会出现，请问这是因为面包板的硬件不支持realtime吗

目前AEC需要硬件支持。

Alex-toughman, 2025-04-07:
> > sorry 刚才看到如果borad type选择ESP BoX3那个选项就会出现，请问这是因为面包板的硬件不支持realtime吗
> 
> 目前AEC需要硬件支持。

请问虾哥，面包板有自己安装aec的教程吗？谢谢

78, 2025-04-07:
需要硬件回采设计，我感觉买这样的模块的成本，还不如买一个立创开发板S3.

closed 2025-04-09 by 78