78/xiaozhi-esp32 #452: 上传列表数据

HelloDB, 2025-04-08:
现在的模型iot感觉理解音乐类指令时表现特别差，Iot属性定义了："歌曲:匆匆那年 url:http://node2333.com/music/ccnn.mp3"或 "歌曲:匆匆那年 /music/ccnn.mp3"，指令中让他返回完整信息或提取url或文件路径用来播放，他总是自己凭空创造，类似：http://local_music_library_url/rush_year.mp3的无效地址。
这套方法控制米家设备时却能准确理解并返回设备信息完成控制，即使属性里定义了几十行设备信息都能有95%的准确率。
MCP接口不知道有大致的时间节点没？
目前有什么办法把用户自己歌曲库列表准确的传递给用户自己创建的“智能体”吗？
硬件端、小智控制面板、或者api方式都行，有这个途径应该就能很好的扩展小智音乐能力了。

78, 2025-04-13:
> 现在的模型iot感觉理解音乐类指令时表现特别差，Iot属性定义了："歌曲:匆匆那年 url:[http://node2333.com/music/ccnn.mp3"或](http://node2333.com/music/ccnn.mp3%22%E6%88%96) "歌曲:匆匆那年 [...]

MCP月底部分用户开放，下个月全部开放。

Funz-001, 2025-08-02:
@78 Has there been any update to support music MCP yet?