78/xiaozhi-esp32 #453: 如果小智ai能嵌入到各种遥控器，做成智能遥控器，用语音代替按键，那将会非常炫酷，但它的待机功耗是个问题！

wyk5656, 2025-04-08:
### Is your feature request related to a problem?

在保留语音唤醒的前提下，我不知道小智ai最低待机功耗能做到多少？虾哥您是否有测试过？

### Describe the solution you'd like.

如果我们让ESP32及各部分模块进入睡眠模式，然后通过设置音频处理模块以低功耗模式侦测到语音信号时通过中断唤醒ESP32，亦或者使用专门的语音唤醒算法来处理麦克风输入的数据，当检测到目标音频（例如“嘿，ESP32”）时，ESP32就会从深度睡眠中醒来，以此来实现超低待机功耗，当然这些都只是我的猜想，我不知道这样能否将待机电流做到uA级？
希望虾哥能给予一些意见和看法！

78, 2025-04-08:
有专门的低功耗唤醒芯片，肯定是个位数mA的电流

meibao-real, 2025-04-10:
关注到了这个项目，正好看到这个issue，我正在做低功耗版本的，等我测试好就在立创广场发布开源，乐鑫本来就有低功耗模式，在无动作时自动进入低功耗模式，关闭高速晶振切换到外部低速32.768晶振继续保持运行，理论上整体电流500个微安左右（通过MOS管关闭所有外设供电）

wyk5656, 2025-04-10:
> 关注到了这个项目，正好看到这个issue，我正在做低功耗版本的，等我测试好就在立创广场发布开源，乐鑫本来就有低功耗模式，在无动作时自动进入低功耗模式，关闭高速晶振切换到外部低速32.768晶振继续保持运行，理论上整体电流500个微安左右（通过MOS管关闭所有外设供电）

closed 2025-04-10 by wyk5656

78, 2025-04-10:
> 关注到了这个项目，正好看到这个issue，我正在做低功耗版本的，等我测试好就在立创广场发布开源，乐鑫本来就有低功耗模式，在无动作时自动进入低功耗模式，关闭高速晶振切换到外部低速32.768晶振继续保持运行，理论上整体电流500个微安左右（通过MOS管关闭所有外设供电）

这样wifi基本上就断掉了

meibao-real, 2025-04-10:
> > 关注到了这个项目，正好看到这个issue，我正在做低功耗版本的，等我测试好就在立创广场发布开源，乐鑫本来就有低功耗模式，在无动作时自动进入低功耗模式，关闭高速晶振切换到外部低速32.768晶振继续保持运行，理论上整体电流500个微安左右（通过MOS管关闭所有外设供电）
> 
> 这样wifi基本上就断掉了

不会，WiFi和BLE都可以保持连接不断

78, 2025-04-10:
> > > 关注到了这个项目，正好看到这个issue，我正在做低功耗版本的，等我测试好就在立创广场发布开源，乐鑫本来就有低功耗模式，在无动作时自动进入低功耗模式，关闭高速晶振切换到外部低速32.768晶振继续保持运行，理论上整体电流500个微安左右（通过MOS管关闭所有外设供电）
> > 
> > 
> > 这样wifi基本上就断掉了
> 
> 不会，WiFi和BLE都可以保持连接不断

这么强，如果还想维持一个mqtt连接呢？大概90秒一个心跳包。

meibao-real, 2025-04-10:
> > > > 关注到了这个项目，正好看到这个issue，我正在做低功耗版本的，等我测试好就在立创广场发布开源，乐鑫本来就有低功耗模式，在无动作时自动进入低功耗模式，关闭高速晶振切换到外部低速32.768晶振继续保持运行，理论上整体电流500个微安左右（通过MOS管关闭所有外设供电）
> > > 
> > > 
> > > 这样wifi基本上就断掉了
> > 
> > 
> > 不会，WiFi和BLE都可以保持连接不断
> 
> 这么强，如果还想维持一个mqtt连接呢？大概90秒一个心跳包。

没有问题，不过我还在调试，之前签保密协议看过乐鑫官方做的ESP32-C3单火开关项目整套源码，IOT平台用的是腾讯云，协议是MQTT，心跳是60秒，没有经过进一步优化的情况下电流是800微安，保持WIFI连接和MQTT连接的情况下，可以正常等待接收MQTT数据包过来，我现在调试ESP32S3的这个板子功耗还有900多微安，优化目标是500微安左右，还有很多需要改进的地方

meibao-real, 2025-04-10:
> 不知道怎么做的，我不开WiFi，按照文档操作，打开了PM开关，C3还是有20多mA的电流。不知道是操作方式还是测量方式不对。

等我搞好你来研究研究我的源码还有硬件原理图，还有测试电流时要移除所有外设包括LDO，还有三极管等等，还有IO口如果有下拉到GND的也会影响电流，低功耗这个是硬件和软件互相配合才能实现的

78, 2025-04-10:
确实需要花时间专门去研究，期待分享。

MyIsaRookie, 2025-04-11:
> > 不知道怎么做的，我不开WiFi，按照文档操作，打开了PM开关，C3还是有20多mA的电流。不知道是操作方式还是测量方式不对。
> 
> 等我搞好你来研究研究我的源码还有硬件原理图，还有测试电流时要移除所有外设包括LDO，还有三极管等等，还有IO口如果有下拉到GND的也会影响电流，低功耗这个是硬件和软件互相配合才能实现的

期待大佬实现，大佬开源了通知一下。

MyIsaRookie, 2025-04-12:
> 确实需要花时间专门去研究，期待分享。

大佬，我想给小智加入wifi的低功耗功能，所以我想在managed_components的78-esp-wifi-connect文件夹下的wifi_station.cc中的wifi配置中加入wifi的低功耗配置，但是提示找不到esp_pm.h这个头文件，这些不是官方组件中的文件吗，为什么会找不到？如果我想给wifi配置改为低功耗那种要怎么做。大佬看到了有空可以解答一下吗，谢谢了。或者大佬有做好的低功耗的版本吗？

MyIsaRookie, 2025-04-12:
> 确实需要花时间专门去研究，期待分享。

知道为什么找不到esp_pm.h了，使用esp_pm.h要加入电源管理组件，忘记在txt加入电源管理组件了。