78/xiaozhi-esp32 #461: 源码编译的固件 在使用wss 模式时 无法唤醒 也无法 说话

danxinyi, 2025-04-09:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

源码编译 1.5.6

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Power Supply used.

USB

### What is the expected behavior?

识别 唤醒语音指令  识别交互语音指令

### What is the actual behavior?

说唤醒词 无法唤醒  说话也无法识别

### Steps to reproduce.

开发板 选择 (X) 无名科技星智1.54(ML307)      
    Flash size (8 MB)  --->    编译    烧录  之后 屏幕连网都没问题   无法识别唤醒词   按键聆听之后 无法识别语音  也没有显示任何 语音文字 一直处于聆听中  几分钟后会出现 再见的流程 说出服务器传出的再见的语音 然后处于 待命状态

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378d03
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x568
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc18
load:0x403cb700,len:0x2df8
entry 0x403c8904
I (36) octal_psram: vendor id    : 0x0d (AP)
I (36) octal_psram: dev id       : 0x02 (generation 3)
I (36) octal_psram: density      : 0x03 (64 Mbit)
I (41) octal_psram: good-die     : 0x01 (Pass)
I (46) octal_psram: Latency      : 0x01 (Fixed)
I (51) octal_psram: VCC          : 0x01 (3V)
I (56) octal_psram: SRF          : 0x01 (Fast Refresh)
I (62) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (68) octal_psram: BurstLen     : 0x01 (32 Byte)
I (73) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (79) octal_psram: DriveStrength: 0x00 (1/1)
I (85) MSPI Timing: PSRAM timing tuning index: 4
I (89) esp_psram: Found 8MB PSRAM device
I (94) esp_psram: Speed: 80MHz
I (98) cpu_start: Multicore app
I (113) cpu_start: Pro cpu start user code
I (113) cpu_start: cpu freq: 240000000 Hz
I (114) app_init: Application information:
I (116) app_init: Project name:     xiaozhi
I (121) app_init: App version:      1.5.6
I (126) app_init: Compile time:     Apr  9 2025 20:41:51
I (132) app_init: ELF file SHA256:  5b4b76e54...
I (137) app_init: ESP-IDF:          v5.3.2-dirty
I (142) efuse_init: Min chip rev:     v0.0
I (147) efuse_init: Max chip rev:     v0.99 
I (152) efuse_init: Chip rev:         v0.2
I (157) heap_init: Initializing. RAM available for dynamic allocation:
I (164) heap_init: At 3FCA9A70 len 0003FCA0 (255 KiB): RAM
I (170) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (176) heap_init: At 600FE108 len 00001ED0 (7 KiB): RTCRAM
I (183) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (191) spi_flash: detected chip: winbond
I (195) spi_flash: flash io: qio
I (200) sleep: Configure to isolate all GPIO pins in sleep state
I (206) sleep: Enable automatic switching of GPIO sleep configuration
I (213) main_task: Started on CPU0
I (223) esp_psram: Reserving pool of 48K of internal memory for DMA/internal allocations
I (223) main_task: Calling app_main()
I (233) BackgroundTask: background_task started
I (243) Board: UUID=9ea07533-abc1-4650-beb2-18aa08380f19 SKU=xingzhi-cube-1.54tft-ml307
I (243) uart: queue free spaces: 100
I (253) button: IoT Button Version: 3.5.0
I (253) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (263) button: IoT Button Version: 3.5.0
I (263) gpio: GPIO[40]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (273) button: IoT Button Version: 3.5.0
I (283) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (293) gpio: GPIO[38]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (303) PowerSaveTimer: Power save timer enabled
I (303) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (313) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (443) Display: Power management not supported
I (483) LcdDisplay: Turning display on
I (483) LcdDisplay: Initialize LVGL library
I (483) LcdDisplay: Initialize LVGL port
I (483) LVGL: Starting LVGL task
I (483) LcdDisplay: Adding LCD screen
I (503) Backlight: Set brightness to 75
I (503) Application: STATE: starting
I (503) NoAudioCodec: Simplex channels created
I (503) Application: ML307 board detected, setting opus encoder complexity to 5
I (513) AudioCodec: Set input enable to true
I (513) AudioCodec: Set output enable to true
I (523) AudioCodec: Audio codec started
I (533) Ml307AtModem: Detecting baud rate...
I (553) Ml307AtModem: Detected baud rate: 921600
I (553) Ml307AtModem: Waiting for network ready...
I (563) Ml307Board: ML307 Module: ML307R-DL-MBRH0S00
I (563) Ml307Board: ML307 IMEI: 460240404208136
I (563) Ml307Board: ML307 ICCID: 898608501224D0285636
I (573) MODEL_LOADER: The storage free size is 21248 KB
I (573) MODEL_LOADER: The partition size is 960 KB
I (573) MODEL_LOADER: Successfully load srmodels
I (583) AFE: AFE Version: (1MIC_V250121)
I (583) AFE: Input PCM Config: total 1 channels(1 microphone, 0 playback), sample rate:16000
I (593) AFE: AFE Pipeline: [input] -> |NS(WebRTC)| -> |VAD(WebRTC)| -> [output]
I (603) AudioProcessor: Audio communication task started, feed size: 160 fetch size: 512
I (613) WakeWordDetect: Model 0: wn9_nihaoxiaozhi_tts
I (623) AFE_CONFIG: Set WakeNet Model: wn9_nihaoxiaozhi_tts
I (653) Ota: Current version: 1.5.6
I (653) Ml307Http: HTTP 连接已创建，ID: 0
MC Quantized wakenet9: wakenet9l_tts1h8_你好小智_3_0.631_0.635, tigger:v3, mode:0, p:0, (Feb 18 2025 12:00:54)
I (663) AFE: AFE Version: (1MIC_V250121)
I (673) AFE: Input PCM Config: total 1 channels(1 microphone, 0 playback), sample rate:16000
I (673) AFE: AFE Pipeline: [input] ->  -> |VAD(WebRTC)| -> |WakeNet(wn9_nihaoxiaozhi_tts,)| -> [output]
I (683) WakeWordDetect: Audio detection task started, feed size: 512 fetch size: 512
I (693) Application: STATE: idle
I (703) main_task: Returned from app_main()
I (1303) PowerSaveTimer: Power save timer disabled
I (1303) PowerManager: ADC value: 2491 average: 2491 level: 100
I (1833) Ml307Http: HTTP请求成功，状态码: 200
I (1853) Ml307Http: HTTP连接已关闭，ID: 0
I (1863) Ota: Current is the latest version
I (1863) Ota: Running partition: app
E (1863) Ota: Failed to get state of partition
I (1873) Application: Resampling audio from 16000 to 24000
I (1873) OpusResampler: Resampler configured with input sample rate 16000 and output sample rate 24000
I (2303) PowerManager: ADC value: 2361 average: 2426 level: 99
I (3303) PowerManager: ADC value: 2352 average: 2401 level: 93
I (10063) Application: STATE: connecting
I (10073) WebSocket: Connecting to wss://api.tenclass.net:443/xiaozhi/v1/
I (10723) Application: STATE: listening
I (20703) Application: Free internal: 48379 minimal internal: 45283
I (30703) Application: Free internal: 52303 minimal internal: 45283
I (40703) Application: Free internal: 52303 minimal internal: 45283
I (50703) Application: Free internal: 50135 minimal internal: 45283
```

### More Information.

开发板选择  无名  1.54  4G

78, 2025-04-09:
之前有这个问题么？

ayi21sui, 2025-04-09:
没遇到过 第一次编译   用他人提供的固件  看日志是使用的udp模式 可以正常说话交流

78, 2025-04-10:
用最新的1.5.8或者main分支的代码试试

ayi21sui, 2025-04-10:
.obj -MF esp-idf\main\CMakeFiles\__idf_main.dir\application.cc.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/application.cc.obj -c D:/python/xiaozhi-esp32/main/application.cc
D:/python/xiaozhi-esp32/main/application.cc: In member function 'void Application::CheckNewVersion()':
D:/python/xiaozhi-esp32/main/application.cc:89:29: error: 'display' was not declared in this scope; did you mean 'Display'?
   89 |             Schedule([this, display]() {
      |                             ^~~~~~~
      |                             Display
D:/python/xiaozhi-esp32/main/application.cc: In lambda function:
D:/python/xiaozhi-esp32/main/application.cc:92:17: error: 'display' is not captured
   92 |                 display->SetIcon(FONT_AWESOME_DOWNLOAD);
      |                 ^~~~~~~
D:/python/xiaozhi-esp32/main/application.cc:89:36: note: the lambda has no capture-default
   89 |             Schedule([this, display]() {
      |                                    ^
D:/python/xiaozhi-esp32/main/application.cc:89:29: note: '<typeprefixerror>display' declared here
   89 |             Schedule([this, display]() {
      |                             ^~~~~~~
D:/python/xiaozhi-esp32/main/application.cc:94:17: error: 'display' is not captured
   94 |                 display->SetChatMessage("system", message.c_str());
      |                 ^~~~~~~
D:/python/xiaozhi-esp32/main/application.cc:89:36: note: the lambda has no capture-default
   89 |             Schedule([this, display]() {
      |                                    ^
D:/python/xiaozhi-esp32/main/application.cc:89:29: note: '<typeprefixerror>display' declared here
   89 |             Schedule([this, display]() {
      |                             ^~~~~~~
D:/python/xiaozhi-esp32/main/application.cc:114:36: error: 'display' is not captured
  114 |                 ota_.StartUpgrade([display](int progress, size_t speed) {
      |                                    ^~~~~~~
D:/python/xiaozhi-esp32/main/application.cc:89:36: note: the lambda has no capture-default
   89 |             Schedule([this, display]() {
      |                                    ^
D:/python/xiaozhi-esp32/main/application.cc:89:29: note: '<typeprefixerror>display' declared here
   89 |             Schedule([this, display]() {
      |                             ^~~~~~~
D:/python/xiaozhi-esp32/main/application.cc: In lambda function:
D:/python/xiaozhi-esp32/main/application.cc:117:21: error: 'display' is not captured
  117 |                     display->SetChatMessage("system", buffer);
      |                     ^~~~~~~
D:/python/xiaozhi-esp32/main/application.cc:114:43: note: the lambda has no capture-default
  114 |                 ota_.StartUpgrade([display](int progress, size_t speed) {
      |                                           ^
D:/python/xiaozhi-esp32/main/application.cc:89:29: note: '<typeprefixerror>display' declared here
   89 |             Schedule([this, display]() {
      |                             ^~~~~~~
D:/python/xiaozhi-esp32/main/application.cc: In lambda function:
D:/python/xiaozhi-esp32/main/application.cc:121:17: error: 'display' is not captured
  121 |                 display->SetStatus(Lang::Strings::UPGRADE_FAILED);
      |                 ^~~~~~~
D:/python/xiaozhi-esp32/main/application.cc:89:36: note: the lambda has no capture-default
   89 |             Schedule([this, display]() {
      |                                    ^
D:/python/xiaozhi-esp32/main/application.cc:89:29: note: '<typeprefixerror>display' declared here
   89 |             Schedule([this, display]() {
      |                             ^~~~~~~
D:/python/xiaozhi-esp32/main/application.cc: In member function 'void Application::CheckNewVersion()':
D:/python/xiaozhi-esp32/main/application.cc:133:9: error: 'display' was not declared in this scope; did you mean 'Display'?
  133 |         display->ShowNotification(message.c_str());
      |         ^~~~~~~
      |         Display
[1870/1877] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/boards/common/ml307_board.cc.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the D:\python\xiaozhi-esp32\build\log\idf_py_stderr_output_22748 and D:\python\xiaozhi-esp32\build\log\idf_py_stdout_output_22748
PS D:\python\xiaozhi-esp32>
1.58 无法构建

ayi21sui, 2025-04-10:
1.5.8 已编译 和 上面1.5.6 一样的情况 无法说话

78, 2025-04-11:
> 1.5.8 已编译 和 上面1.5.6 一样的情况 无法说话

目前没有收到其他用户反馈有类似问题