78/xiaozhi-esp32 #463: 将Button组件从V3.3.1升级到V4.1.2

SuhaylZhao, 2025-04-10:
### Is your feature request related to a problem?

大佬们好， 小智项目目前使用的Button组件版本是V3.3.1 [espressif/button: "^3.3.1"](https://github.com/78/xiaozhi-esp32/blob/main/main/idf_component.yml#L18) ,  该版本GPIO Button没啥问题，ADC Button有较大缺陷，默认使用 [ADC_UNIT_1](https://github.com/espressif/esp-iot-solution/blob/b860c6933cf05df0a833022219183855bf0e3f9b/components/button/button_adc.c#L47) ，无法使用ADC_UNIT_2，即使在代码中改了也没用。在最新的版本[ V4.1.2](https://components.espressif.com/components/espressif/button/versions/4.1.2?language=en) 中，允许用户选择使用ADC1或者ADC2，适用性更强。

### Describe the solution you'd like.

将Button组件由V3.3.1升级到V4.1.2, 不过这两个版本API变化较大，需要修改小智项目中的封装好的Button类， 不过似乎还影响到了 [espressif__esp_lvgl_port组件](https://github.com/espressif/esp-iot-solution/blob/b860c6933cf05df0a833022219183855bf0e3f9b/components/button/button_adc.c#L47) ，不知道大佬是否有升级该组件的计划，期待大佬更新

laride, 2025-04-10:
> 不过似乎还影响到了 [espressif__esp_lvgl_port组件](https://github.com/espressif/esp-iot-solution/blob/b860c6933cf05df0a833022219183855bf0e3f9b/components/button/button_adc.c#L47)

[esp_lvgl_port 2.5.0](https://components.espressif.com/components/espressif/esp_lvgl_port/versions/2.5.0) 以上版本已适配 button v4，详见 esp_lvgl_port [CHANGELOG](https://github.com/espressif/esp-bsp/blob/master/components/esp_lvgl_port/CHANGELOG.md#features-functional-change-for-button-v4-users)

78, 2025-04-10:
可以，麻烦测试成功了提个PR

closed 2025-05-20 by 78