78/xiaozhi-esp32 #468: 编译完成无法烧录，

wsdmzz, 2025-04-10:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

1.5.6

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

烧录成功运行

### What is the actual behavior?

按键进入烧录模式，实际显示

[image]

### Steps to reproduce.

1. Step  克隆仓库
2. Step  更改"ws://192.168.1.25:8000“
3. Step  编译
4. Step   按键进入烧录模式
5. Step    进行烧录  没有烧录进度
...

### Build or installation Logs.

```plain

```

yingfuemi96, 2025-04-10:
我其实有些好奇这个版本你怎么编译过的？我拉取之后提示display编译报错

layola13, 2025-04-16:
代码命令行烧 ，编译后用flash指令烧入。