78/xiaozhi-esp32 #479: 我有一台宇树G1可二次开发版，有没有合作意向将小智AI接入G1的实践，一起学习技术实践

Brave-AirPig, 2025-04-12:
### Is your feature request related to a problem?

可以联系我vx:ttkx3285

layola13, 2025-05-05:
可以哇，不错，高大上，我现在正在重构小智esp32 C++底层，改掉2选1协议的尴尬，支持http+mqtt+websocket 灵活调用，

huangjunsen0406, 2025-09-04:
> ### Is your feature request related to a problem?
> 可以联系我vx:ttkx3285
> 
> ### Describe the solution you'd like.
> _No response_
> 
> ### Describe alternatives you've considered.
> _No response_
> 
> ### Additional context.
> _No response_

可以看看我的python版小智，支持在linux嵌入式设备部署，目前有网友在宇树其他机型测试过了，还有在机器狗上 [py-xiaozhi](https://github.com/huangjunsen0406/py-xiaozhi.git)