78/xiaozhi-esp32 #481: c3 - 聆听对话状态，多轮对话后随机重启到待机状态

sidaotiger, 2025-04-13:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

ex.1.5.8_xmini_c3

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

PowerShell

### Power Supply used.

USB

### What is the expected behavior?

abort() was called at PC 0x4213068d on core 0 引起重启
重启日志见附件

### What is the actual behavior?

重启到待机状态

### Steps to reproduce.

1. 对话
2. 聆听和多轮对话
3. 重启到待机状态
4. 串口监控发现abort
5. 多次复现

### Debug Logs.

```plain
I (125519) Application: Abort speaking
I (125739) Application: STATE: listening

abort() was called at PC 0x4213068d on core 0
Core  0 register dump:
MEPC    : 0x40380880  RA      : 0x40387b44  SP      : 0x3fca99a0  GP      : 0x3fc91e00  
TP      : 0x3fca9bd0  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000004  S1      : 0x3fca9a04  A0      : 0x3fca99cc  A1      : 0x3fca9a02  
A2      : 0x00000000  A3      : 0x3fca99f9  A4      : 0x00000001  A5      : 0x3fc9b000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fca9ae0  S3      : 0x3fcb7938  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fca99a0: 0x00000000 0x00000000 0x3fca9a00 0x4038fb40 0x00000000 0x00000000 0x3fc9c468 0x3fc93660
3fca99c0: 0x3fca9a04 0x3fc9367c 0x3fca9a00 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
3fca99e0: 0x43502074 0x34783020 0x30333132 0x20643836 0x63206e6f 0x2065726f 0x00000030 0x42000000
3fca9a00: 0x00000030 0x33313234 0x64383630 0x4212fa00 0x00000000 0x000005dc 0x3fcd3c30 0x42130690
3fca9a20: 0x3fca9ae0 0x000005dc 0x3fcd3c30 0x421306dc 0x3fcd3c30 0x000005dc 0x3fcd3c30 0x42130802
3fca9a40: 0x3fca1a18 0x000005dc 0x4214989a 0x3c22f70c 0x3fc9eae8 0x000005dc 0x000005dc 0x421308ea
3fca9a60: 0x3fca9ae0 0xffffffff 0x3fca9aac 0x42011c48 0x00000000 0x00000000 0x00000000 0x42012a66
3fca9a80: 0x3fca9ae0 0x000005dc 0x3fca9aac 0x4202379e 0x3fca9b74 0x3fc9f044 0x3fcb7924 0x420238ce
3fca9aa0: 0x00000000 0x3fca9b74 0x3fc9f040 0x00000000 0x00000000 0x00000000 0x3fcb7924 0x4038fcd2
3fca9ac0: 0x00000000 0x00000000 0x3fca9ac8 0x00000000 0x00000000 0x3fca1a18 0x3fc9ec98 0x42010084
3fca9ae0: 0x3fc95400 0x00000000 0x4200fb1a 0x4201258a 0x00000000 0x00000000 0x3fc9ec98 0x420100ae
3fca9b00: 0x00000000 0x00000001 0x00000000 0x4200a210 0x00000000 0xffffffff 0x3fca1a02 0x42014f3e
3fca9b20: 0x00000000 0x3fca9ac8 0x00000000 0x00000000 0x00000000 0x3fca1a18 0x3fc9f090 0x42014f9a
3fca9b40: 0x00000000 0x3fca1a18 0x3fca9b6c 0x4200a210 0x00000000 0x3fca1a18 0x3fca1a28 0x4201519e
3fca9b60: 0x00000000 0x00000000 0x00000000 0x3fca1a18 0x00000000 0x3fc9f090 0x3fc9f090 0x00000001
3fca9b80: 0x00000000 0x00000000 0x00000000 0x4201520c 0x00000000 0x00000000 0x00000000 0x40388bfc
3fca9ba0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9bc0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000160 0x3fca9430 0x00003162
3fca9be0: 0x3fc94b40 0x3fc94b40 0x3fca9bd8 0x3fc94b38 0x00000017 0x3fca9af4 0x3fca9af4 0x3fca9bd8
3fca9c00: 0x00000000 0x00000002 0x3fca1bd4 0x6b636162 0x756f7267 0x745f646e 0x006b7361 0x3fca9bd0
3fca9c20: 0x00000006 0x00000000 0x00000002 0x00000001 0x3fcb79ec 0x42007538 0x01a905ed 0x00000000
3fca9c40: 0x3fc9bfa8 0x3fc9c010 0x3fc9c078 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
3fca9c60: 0x00000000 0x42006106 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9c80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9ca0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9cc0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9ce0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9d00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9d20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fca9bd0 0x00000020 0x07f33461
3fca9d40: 0x00000000 0x000f4240 0x02000000 0x4201251a 0x3fc95400 0x3fccf6d0 0x3fc9df74 0x00000090
3fca9d60: 0x3c1b0d14 0x3fca9e24 0x00000024 0x00000025 0x1eb2055c 0xfecabd04 0xe30d9e49 0xf6ec7600
3fca9d80: 0x3fca9ee4 0x3fcaa3ec 0x3fcaa45c 0x3fcaa4a8 0x00000009 0x3fca9e50 0x3fca9d60 0x00000000

ELF file SHA256: dba10a737

Rebooting...
ESP-ROM:esp32c3-api1-20210207
```

78, 2025-04-13:
试试 1.5.9 https://github.com/78/xiaozhi-esp32/releases/tag/v1.5.9

78, 2025-04-15:
测试一下 1.6.0 版本的 xmini-c3

winslow49, 2025-04-17:
我测了1.6版本还有问题

78, 2025-04-17:
> 我测了1.6版本还有问题

为什么呢

jyoryo, 2025-05-26:
你好，我升级到1.6.4还会出现这个问题。
XiaoZhi AI firmware version.
xmini_c3

Operating System used.
MacOS

How did you build your project?
VS Code ESF-IDF运行构建、烧录和监视

Power Supply used.
USB

What is the expected behavior?
abort() was called at PC 0x42131e59 on core 0

What is the actual behavior?
重启到待机状态

Steps to reproduce.
对话
聆听和多轮对话
重启到待机状态
串口监控发现abort
多次复现
Debug Logs.
`
I (7290507) Application: Free internal: 69116 minimal internal: 59484
I (7300507) Application: Free internal: 68932 minimal internal: 59484
I (7310507) Application: Free internal: 68768 minimal internal: 59484
I (7313937) Application: STATE: listening

abort() was called at PC 0x42131e59 on core 0
--- 0x42131e59: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin21.1/riscv32-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:45

Core  0 register dump:
MEPC    : 0x40380822  RA      : 0x403878da  SP      : 0x3fca97f0  GP      : 0x3fc91c00  
--- 0x40380822: panic_abort at /Users/jyoryo/work/resources/esp/v5.3.1/esp-idf/components/esp_system/panic.c:463
--- 0x403878da: __ubsan_include at /Users/jyoryo/work/resources/esp/v5.3.1/esp-idf/components/esp_system/ubsan.c:311

TP      : 0x3fca9e30  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000004  S1      : 0x3fca9854  A0      : 0x3fca981c  A1      : 0x3fca9852  
A2      : 0x00000000  A3      : 0x3fca9849  A4      : 0x00000001  A5      : 0x3fc9b000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fcdea88  S3      : 0x000005a0  
S4      : 0x3fcde4e4  S5      : 0x3fcde4e4  S6      : 0x00000000  S7      : 0x3fcddf44  
S8      : 0x000002d0  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
--- 0x40380001: _vector_table at /Users/jyoryo/work/resources/esp/v5.3.1/esp-idf/components/riscv/vectors_intc.S:54

MHARTID : 0x00000000  

Stack memory:
3fca97f0: 0x00000000 0x00000000 0x3fca9850 0x4038f87e 0x00000000 0x00000000 0x3fc9c640 0x3fc934b4
--- 0x4038f87e: abort at /Users/jyoryo/work/resources/esp/v5.3.1/esp-idf/components/newlib/abort.c:35

3fca9810: 0x3fca9854 0x3fc934d0 0x3fca9850 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
3fca9830: 0x43502074 0x34783020 0x31333132 0x20393565 0x63206e6f 0x2065726f 0x00000030 0x42000000
3fca9850: 0x00000030 0x33313234 0x39356531 0x42131300 0x000
--- 0x42131300: __gnu_cxx::__mutex::unlock() at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin21.1/riscv32-esp-elf/build/build-cc-gcc-final/riscv32-esp-elf/rv32imc_zicsr_zifencei/ilp32/no-rtti/libstdc++-v3/include/ext/concurrence.h:155

00000 0x3fcde4e8 0x3fcdbce4 0x42131e5c
--- 0x42131e5c: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin21.1/riscv32-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:49

3fca9870: 0x3fcdea88 0x3fcde4e8 0x3fcdbce4 0x42131ea8 0x3fcdbce4 0x3fcde4e8 0x3fcdbce4 0x42131fce
--- 0x42131ea8: __cxxabiv1::__unexpected(void (*)()) at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin21.1/riscv32-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:64
--- 0x42131fce: __cxa_rethrow at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin21.1/riscv32-esp-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:103

3fca9890: 0x3fca9d40 0x3fcde4e8 0x4214f4d8 0x3c22aad0 0x3fca9d40 0x3fcde4e8 0x00000b40 0x421320b6
--- 0x4214f4d8: std::bad_alloc::~bad_alloc() at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin21.1/riscv32-esp-elf/src/gcc/libstdc++-v3/libsupc++/bad_alloc.cc:28
[... 32 lines cut ...]
3fca9af0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9b10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9b30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9b50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9b70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9b90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9bb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9bd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 01049502b

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xe (SPI_FAST_FLASH_BOOT)
Saved PC:0x403807da
--- 0x403807da: esp_restart_noos at /Users/jyoryo/work/resources/esp/v5.3.1/esp-idf/components/esp_system/port/soc/esp32c3/system_internal.c:110

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x3d0
load:0x403cc710,len:0x96c
load:0x403ce710,len:0x2c24
entry 0x403cc710
`