78/xiaozhi-esp32 #483: audio_codec增加直接输入输出opus的支持

HoGC, 2025-04-14:
### Is your feature request related to a problem?

目前有许多采用独立的语音芯片方案，有些有opus的编解码能力，可以大大的降低esp内存和cpu的消耗
如四博智联的AI-02，使用的语音芯片有opus的编码能力，完全可以使用audio_codec进行编码

### Describe the solution you'd like.

目前的audio_codec没有直接输出opus的抽象接口，我尝试增加以下接口
```
#ifdef CONFIG_USE_AUDIO_CODEC_DECODE_OPUS
    virtual void OutputData(std::vector<uint8_t>& opus);
#endif
#ifdef CONFIG_USE_AUDIO_CODEC_ENCODE_OPUS
    virtual bool InputData(std::vector<uint8_t>& opus);
#endif

#ifdef CONFIG_USE_AUDIO_CODEC_DECODE_OPUS
    virtual bool ConfigDecode(int sample_rate, int channels, int duration_ms);
#endif

#ifdef CONFIG_USE_AUDIO_CODEC_ENCODE_OPUS
    virtual bool ConfigEncode(int sample_rate, int channels, int duration_ms);
#endif
```
你觉得加到main上合适吗，或者你有更好的实现

78, 2025-04-14:
可以的，或者抽象一下 Encoder 和 Decoder 的接口。可以先做出来看看有没有问题

HoGC, 2025-04-14:
AI-01的仓库有抽象过Encoder 和 Decoder，改动较大不方便合并，后面又改回来
抽象Encoder 和 Decoder ，也还是要在audio_codec实现read uint8的类型，
目前你看下AI-01的仓库的实现，是没有抽象Encoder 和 Decoder的实现，你觉得可以的话，我整理下提一个pr

HoGC, 2025-04-14:
https://github.com/SmartArduino/DOIT_AI
以下是其中一处的修改，对AUDIO_CODEC_ENCODE_OPUS单独处理
```
    if (device_state_ == kDeviceStateListening) {
#ifdef CONFIG_USE_AUDIO_CODEC_ENCODE_OPUS
        std::vector<uint8_t> opus;
        ReadAudio(opus, 16000, 30 * 16000 / 1000);
        Schedule([this, opus = std::move(opus)]() {
            protocol_->SendAudio(opus);
        });
#else
        ReadAudio(data, 16000, 30 * 16000 / 1000);
        background_task_->Schedule([this, data = std::move(data)]() mutable {
            opus_encoder_->Encode(std::move(data), [this](std::vector<uint8_t>&& opus) {
                Schedule([this, opus = std::move(opus)]() {
                    protocol_->SendAudio(opus);
                });
            });
        });
#endif
```

HoGC, 2025-04-14:
这是之前抽象opus_codecs的
https://github.com/SmartArduino/DOIT_AI/tree/46f36d731146999db3e7e88ccc91919182d1bbe0/main/opus_codecs

78, 2025-04-14:
这看起来改动也不大呢

78, 2025-04-15:
> 抽象opus_codecs的话application改动多点，主要是no_opus_codecs解码后的数据定义为opus也不是pcm也不是，然后终归需要直接输入输出opus的接口，感觉在audio_codec实现也是合适的，那确实好像也不用抽象出来，要不先提一版没有opus_codecs的，你看看

支持