78/xiaozhi-esp32 #485: lilygo-t-watch 2020 v3

macauhome, 2025-04-14:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

v1.2.1 bread-compact-wifi

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

PowerShell

### Power Supply used.

USB

### What is the expected behavior?

在lilygo-t-watch 2020 v3 上運行小智AI

### What is the actual behavior?

報行時手錶没顯示, 在ESP-IDF Monitor 顯示錯誤信息如下, 
E (209) i2c.master: I2C hardware NACK detected
E (209) i2c.master: I2C transaction unexpected nack detected
E (219) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed 
E (229) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
E (229) lcd_panel.io.i2c: panel_io_i2c_tx_buffer(193): i2c transaction failed  
E (239) lcd_panel.ssd1306: panel_ssd1306_init(151): io tx param SSD1306_CMD_SET_MULTIPLEX failed
E (249) CompactWifiBoard: Failed to initialize display

我的config.h file內容如下
`#ifndef _BOARD_CONFIG_H_
#define _BOARD_CONFIG_H_
#include <driver/gpio.h>

// ======== 音频配置 ========
// MAX98357A I2S音频放大器
#define AUDIO_INPUT_SAMPLE_RATE 16000
#define AUDIO_OUTPUT_SAMPLE_RATE 16000
#define AUDIO_I2S_GPIO_WS      GPIO_NUM_25  // I2S_WS (LRCK)
#define AUDIO_I2S_GPIO_BCLK    GPIO_NUM_26  // I2S_BCK (SCK)
#define AUDIO_I2S_GPIO_DOUT    GPIO_NUM_33  // I2S_DOUT (DATA)

// ======== 显示屏配置 ========
// ST7789V SPI显示屏
#define DISPLAY_SPI_SCK_PIN    GPIO_NUM_18  // TFT_SCLK
#define DISPLAY_SPI_MOSI_PIN   GPIO_NUM_19  // TFT_MOSI
#define DISPLAY_DC_PIN         GPIO_NUM_27  // TFT_DC
#define DISPLAY_SPI_CS_PIN     GPIO_NUM_5   // TFT_CS
#define DISPLAY_BACKLIGHT_PIN  GPIO_NUM_12  // TFT_BL
#define DISPLAY_WIDTH          240
#define DISPLAY_HEIGHT         240
#define DISPLAY_BACKLIGHT_OUTPUT_INVERT false

// ======== 传感器配置 ========
// BMA423加速度计 (I2C)
#define BMA423_SDA             GPIO_NUM_21
#define BMA423_SCL             GPIO_NUM_22
#define BMA423_INT             GPIO_NUM_39

// ======== 电源管理 ========
// AXP202 PMIC (I2C)
#define AXP202_SDA             GPIO_NUM_21  // 与BMA423共享I2C总线
#define AXP202_SCL             GPIO_NUM_22
#define AXP202_INT             GPIO_NUM_35   // 电池中断

// ======== 触摸屏配置 ========
// FT6336U 触摸控制器 (I2C)
#define TOUCH_I2C_SDA_PIN      GPIO_NUM_23
#define TOUCH_I2C_SCL_PIN      GPIO_NUM_32
#define TOUCH_I2C_ADDR         0x38
#define TOUCH_INT_PIN          GPIO_NUM_36   // 注意与BOOT按钮冲突

// ======== 其他外设 ========
#define MOTOR_PIN              GPIO_NUM_4   // 震动电机
#define IR_PIN                 GPIO_NUM_13  // 红外发射
#define RTC_PIN                GPIO_NUM_37  // RTC中断
#define BOOT_BUTTON_GPIO       GPIO_NUM_36  // 与触摸中断共享GPIO

#endif // _BOARD_CONFIG_H_

[config.json](https://github.com/user-attachments/files/19732360/config.json)
[partitions_t-watch2020-v3.csv](https://github.com/user-attachments/files/19732361/partitions_t-watch2020-v3.csv)

`

### Steps to reproduce.

1. release.py 
2. flash
3. open esp-idf monitor
...

### Debug Logs.

```plain

```

78, 2025-04-15:
你的日志显示的是 ssd1306 的显示屏，而不是 st7789 的显示屏，检查你的配置是否生效，编译的开发板是否选择正确。

macauhome, 2025-04-15:
> 你的日志显示的是 ssd1306 的显示屏，而不是 st7789 的显示屏，检查你的配置是否生效，编译的开发板是否选择正确。

請問如何配置st7789 ? 謝謝

78, 2025-04-15:
参考其他开发板目录的代码，基本上大部分都是st7789

macauhome, 2025-04-28:
修改了, 但還是没顯示
config.h
`#ifndef BOARD_CONFIG_H
#define BOARD_CONFIG_H

#include <driver/gpio.h>

// ======== 音頻配置 ========
// MAX98357A I2S音頻放大器
#define AUDIO_INPUT_SAMPLE_RATE 16000
#define AUDIO_OUTPUT_SAMPLE_RATE 16000
#define AUDIO_I2S_GPIO_WS GPIO_NUM_25  // I2S_WS (LRCK)
#define AUDIO_I2S_GPIO_BCLK GPIO_NUM_26 // I2S_BCK (SCK)
#define AUDIO_I2S_GPIO_DOUT GPIO_NUM_33 // I2S_DOUT (DATA)
#define AUDIO_I2S_GPIO_DIN GPIO_NUM_34  // Change from GPIO 7 to GPIO 34

// ======== 顯示屏配置 ========
// ST7789V SPI顯示屏
#define DISPLAY_SPI_SCK_PIN GPIO_NUM_18  // TFT_SCLK
#define DISPLAY_SPI_MOSI_PIN GPIO_NUM_19 // TFT_MOSI
#define DISPLAY_DC_PIN GPIO_NUM_27       // TFT_DC
#define DISPLAY_SPI_CS_PIN GPIO_NUM_5    // TFT_CS
#define DISPLAY_BACKLIGHT_PIN GPIO_NUM_12 // TFT_BL
#define DISPLAY_WIDTH 240
#define DISPLAY_HEIGHT 240
//#define DISPLAY_BACKLIGHT_OUTPUT_INVERT false

#define DISPLAY_MIRROR_X true
#define DISPLAY_MIRROR_Y false
#define DISPLAY_SWAP_XY true
#define DISPLAY_OFFSET_X  0
#define DISPLAY_OFFSET_Y  0
#define DISPLAY_BACKLIGHT_PIN GPIO_NUM_42
#define DISPLAY_BACKLIGHT_OUTPUT_INVERT true

// ======== 傳感器配置 ========
// BMA423加速度計 (I2C)
#define BMA423_SDA GPIO_NUM_21
#define BMA423_SCL GPIO_NUM_22
#define BMA423_INT GPIO_NUM_39

// ======== 電源管理 ========
// AXP202 PMIC (I2C)
#define AXP202_SDA GPIO_NUM_21  // 與BMA423共享I2C總線
#define AXP202_SCL GPIO_NUM_22
#define AXP202_INT GPIO_NUM_35  // 電池中斷

// ======== 觸摸屏配置 ========
// FT6336U 觸摸控制器 (I2C)
#define TOUCH_I2C_SDA_PIN GPIO_NUM_23
#define TOUCH_I2C_SCL_PIN GPIO_NUM_32
#define TOUCH_I2C_ADDR 0x38
#define TOUCH_INT_PIN GPIO_NUM_38 //36 // 注意與BOOT按鈕衝突
#define TOUCH_I2C_RST_PIN GPIO_NUM_14

// ======== 其他外設 ========
#define MOTOR_PIN GPIO_NUM_4  // 震動電機
#define IR_PIN GPIO_NUM_13    // 紅外發射
#define RTC_PIN GPIO_NUM_37   // RTC中斷
#define BOOT_BUTTON_GPIO GPIO_NUM_36 // 與觸摸中斷共享GPIO

#endif // BOARD_CONFIG_H
`
lilygo-t-watch.cc
`#include "wifi_board.h"
#include "audio_codecs/es8311_audio_codec.h"
#include "display/lcd_display.h"
#include "application.h"
#include "button.h"
#include "config.h"
[... 196 lines cut ...]
    virtual AudioCodec* GetAudioCodec() override {
        static Es8311AudioCodec audio_codec(
            codec_i2c_bus_, 
            I2C_NUM_0, 
            AUDIO_INPUT_SAMPLE_RATE, 
            AUDIO_OUTPUT_SAMPLE_RATE,
            AUDIO_I2S_GPIO_MCLK, 
            AUDIO_I2S_GPIO_BCLK, 
            AUDIO_I2S_GPIO_WS, 
            AUDIO_I2S_GPIO_DOUT, 
            AUDIO_I2S_GPIO_DIN,
            AUDIO_CODEC_PA_PIN, 
            AUDIO_CODEC_ES8311_ADDR);
        return &audio_codec;
    }

    // 获取显示屏
    virtual Display* GetDisplay() override {
        return display_;
    }
};

// 注册开发板
DECLARE_BOARD(LilygoTWatchBoard);
`