78/xiaozhi-esp32 #488: 1.5.9适配SH1106

HQWYFHJQ, 2025-04-15:
1.5.9适配SH1106时  屏幕除了控件以外区域全部都被填充了颜色 控件颜色为黑色 其余部分被屏幕的显色填充  初始化时发送0XA6可以短暂去掉颜色填充  请问如何关闭除UI控件以外的显示

78, 2025-04-15:
> 1.5.9适配SH1106时 屏幕除了控件以外区域全部都被填充了颜色 控件颜色为黑色 其余部分被屏幕的显色填充 初始化时发送0XA6可以短暂去掉颜色填充 请问如何关闭除UI控件以外的显示

```c
    // draw white
    std::vector<uint16_t> buffer(width_, 0xFFFF);
    for (int y = 0; y < height_; y++) {
        esp_lcd_panel_draw_bitmap(panel_, 0, y, width_, y + 1, buffer.data());
    }
```

这样呢？

HQWYFHJQ, 2025-04-15:
> > 1.5.9 车载SH1106时屏幕除了控件以外区域全部被填充了颜色控件颜色为黑色其余部分被屏幕的显色填充初始化时发送0XA6可以短暂去掉颜色填充请问如何关闭除UI控件之外的显示
> 
>     // draw white
>     std::vector<uint16_t> buffer(width_, 0xFFFF);
>     for (int y = 0; y < height_; y++) {
>         esp_lcd_panel_draw_bitmap(panel_, 0, y, width_, y + 1, buffer.data());
>     }
> 这样呢？

sh1106走的是https://github.com/78/xiaozhi-esp32/blob/main/main/display/oled_display.cc  这里边没有esp_lcd_panel_draw_bitmap函数调用哎

78, 2025-04-15:
找找有没有类似的函数？

kiepphongba, 2025-08-20:
你能给我运行 oled 0.96 的 esp32 s3 n16r8 版本的固件吗？