78/xiaozhi-esp32 #491: 无名星智CUBE1.5.4 TFT 4G版本连接自己的ws服务器没法连接4G

fanyuanxiang, 2025-04-15:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

最新版1.6.0

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Power Supply used.

USB

### What is the expected behavior?

编译成功，但是唤醒时候，会报网络异常.

E (17894) Ml307SslTransport: Failed to connect to xx.xx.xx.xx:10007
E (17894) WebSocket: Failed to connect to server
E (17894) WS: Failed to connect to websocket server
I (17904) Application: STATE: idle
W (17914) Application: Alert 错误: 正在寻找可用服务 [sad]

### What is the actual behavior?

ws 服务器使用的[xiaozhi-esp32-server](https://github.com/xinnan-tech/xiaozhi-esp32-server)

测试连接是正常可以通信的。

[image]

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
(base) PS E:\1_work\100_workspace\1_iot\2_xiaozhi\xiaozhi-esp32> idf.py -p COM9 build flash monitor
Executing action: all (aliases: build)
Running ninja in directory E:\1_work\100_workspace\1_iot\2_xiaozhi\xiaozhi-esp32\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32s3
NOTICE: Skipping optional dependency: tny-robotics/sh1106-esp-idf
NOTICE: Skipping optional dependency: tny-robotics/sh1106-esp-idf
NOTICE: Skipping optional dependency: tny-robotics/sh1106-esp-idf
Processing 25 dependencies:
[25/25] idf (5.3.2)
NOTICE: Skipping optional dependency: tny-robotics/sh1106-esp-idf
-- Project sdkconfig file E:/1_work/100_workspace/1_iot/2_xiaozhi/xiaozhi-esp32/sdkconfig
Loading defaults file E:/1_work/100_workspace/1_iot/2_xiaozhi/xiaozhi-esp32/sdkconfig.defaults...
Loading defaults file E:/1_work/100_workspace/1_iot/2_xiaozhi/xiaozhi-esp32/sdkconfig.defaults.esp32s3...
[... 349 lines cut ...]
I (11024) MODEL_LOADER: The partition size is 960 KB
I (11024) MODEL_LOADER: Successfully load srmodels
I (11034) AFE: AFE Version: (1MIC_V250121)
I (11034) AFE: Input PCM Config: total 1 channels(1 microphone, 0 playback), sample rate:16000
I (11044) AFE: AFE Pipeline: [input] -> |NS(WebRTC)| -> |VAD(WebRTC)| -> [output]
I (11054) AudioProcessor: Audio communication task started, feed size: 160 fetch size: 512
I (11064) WakeWordDetect: Model 0: wn9_nihaoxiaoxin_tts
I (11064) AFE_CONFIG: Set WakeNet Model: wn9_nihaoxiaoxin_tts
MC Quantized wakenet9: wakenet9l_tts1h8_你好小鑫_3_0.635_0.639, tigger:v3, mode:0, p:0, (Apr 10 2025 16:24:24)
I (11104) AFE: AFE Version: (1MIC_V250121)
I (11114) AFE: Input PCM Config: total 1 channels(1 microphone, 0 playback), sample rate:16000
I (11124) AFE: AFE Pipeline: [input] -> |VAD(WebRTC)| -> |WakeNet(wn9_nihaoxiaoxin_tts,)| ->  -> [output]
I (11134) WakeWordDetect: Audio detection task started, feed size: 512 fetch size: 512
I (11144) Application: STATE: idle
I (11154) Application: Resampling audio from 16000 to 24000
I (11154) OpusResampler: Resampler configured with input sample rate 16000 and output sample rate 24000
I (17034) Application: STATE: connecting
I (17094) WebSocket: Connecting to ws://xx.xx.xx.xx:10007/xiaozhi/v1/
I (17454) WakeWordDetect: Encode wake word opus 33 packets in 414 ms
E (17894) Ml307SslTransport: Failed to connect to 106.38.31.47:10007
E (17894) WebSocket: Failed to connect to server
E (17894) WS: Failed to connect to websocket server
I (17904) Application: STATE: idle
W (17914) Application: Alert 错误: 正在寻找可用服务 [sad]
```

78, 2025-04-15:
esp-ml307 缺少对非SSL TCP连接的支持，你需要修改组件的代码，去掉SSL的强制要求

104gogo, 2025-04-16:
https://www.bilibili.com/video/BV1wifwYPEec/?vd_source=55a567379396c008739e5be3fb46e424
4g版本小智连WSS Server攻略

shangyanglj, 2025-04-27:
xiaozhi-esp32\managed_components\78__esp-ml307\ml307_ssl_transport.cc  92行  sprintf(command,"AT+MIPCFG=\"ss1\",%d,1,0",tcp id ); 1改成0

shangyanglj, 2025-04-27:
xiaozhi-esp32\managed_components\78__esp-ml307\ml307_ssl_transport.cc
` sprintf(command,"AT+MIPCFG=\"ssl\",%d,1,0",tcp_id_);`
改为  
`sprintf(command,"AT+MIPCFG=\"ssl\",%d,0,0",tcp_id_);`

78, 2025-04-27:
> xiaozhi-esp32\managed_components\78__esp-ml307\ml307_ssl_transport.cc ` sprintf(command,"AT+MIPCFG=\"ssl\",%d,1,0",tcp_id_);` 改为 `sprintf(command,"AT+MIPCFG=\"ssl\",%d,0,0",tcp_id_);`

只需要改动这行就可以了吗？ 后面可以改成根据url来自动判断。

78, 2025-04-29:
最新的代码已支持根据端口是否为443，自动配置SSL参数。

closed 2025-04-29 by 78