78/xiaozhi-esp32 #495: kevin-sp-v3_board.cc:33:30: error: 'GPIO_NUM_47' was not declared in this scope; did you mean 'GPIO_NUM_7'?

jthou, 2025-04-15:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.6.0, v1.5.4

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

第一次编译，idf.py build之后就出现这个问题。v1.6.0，v1.5.4都能复现。

### What is the actual behavior?

/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc: In member function 'void KEVIN_SP_V3Board::InitializeSpi()':
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc:33:30: error: 'GPIO_NUM_47' was not declared in this scope; did you mean 'GPIO_NUM_7'?
   33 |         buscfg.mosi_io_num = GPIO_NUM_47;
      |                              ^~~~~~~~~~~
      |                              GPIO_NUM_7
In file included from /Users/jintinghou/tmp/GitHub/esp-idf/components/esp_driver_gpio/include/driver/gpio.h:12,
                 from /Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/common/backlight.h:6,
                 from /Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/common/board.h:11,
                 from /Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/common/wifi_board.h:4,
                 from /Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc:1:
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc:39:44: error: 'SPI3_HOST' was not declared in this scope; did you mean 'SPI2_HOST'?
   39 |         ESP_ERROR_CHECK(spi_bus_initialize(SPI3_HOST, &buscfg, SPI_DMA_CH_AUTO));
      |                                            ^~~~~~~~~
/Users/jintinghou/tmp/GitHub/esp-idf/components/esp_common/include/esp_err.h:116:30: note: in definition of macro 'ESP_ERROR_CHECK'
  116 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc: In member function 'void KEVIN_SP_V3Board::InitializeSt7789Display()':
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc:64:33: error: 'GPIO_NUM_45' was not declared in this scope; did you mean 'GPIO_NUM_5'?
   64 |         io_config.dc_gpio_num = GPIO_NUM_45;
      |                                 ^~~~~~~~~~~
      |                                 GPIO_NUM_5
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc:70:50: error: 'SPI3_HOST' was not declared in this scope; did you mean 'SPI2_HOST'?
   70 |         ESP_ERROR_CHECK(esp_lcd_new_panel_io_spi(SPI3_HOST, &io_config, &panel_io));
      |                                                  ^~~~~~~~~
/Users/jintinghou/tmp/GitHub/esp-idf/components/esp_common/include/esp_err.h:116:30: note: in definition of macro 'ESP_ERROR_CHECK'
  116 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
In file included from /Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc:9:
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc: In member function 'virtual Led* KEVIN_SP_V3Board::GetLed()':
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/config.h:20:33: error: 'GPIO_NUM_38' was not declared in this scope; did you mean 'GPIO_NUM_8'?
   20 | #define BUILTIN_LED_GPIO        GPIO_NUM_38
      |                                 ^~~~~~~~~~~
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc:117:30: note: in expansion of macro 'BUILTIN_LED_GPIO'
  117 |         static SingleLed led(BUILTIN_LED_GPIO);
      |                              ^~~~~~~~~~~~~~~~
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc: In member function 'virtual AudioCodec* KEVIN_SP_V3Board::GetAudioCodec()':
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/config.h:16:33: error: 'GPIO_NUM_46' was not declared in this scope; did you mean 'GPIO_NUM_6'?
   16 | #define AUDIO_I2S_SPK_GPIO_BCLK GPIO_NUM_46
      |                                 ^~~~~~~~~~~
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc:123:13: note: in expansion of macro 'AUDIO_I2S_SPK_GPIO_BCLK'
  123 |             AUDIO_I2S_SPK_GPIO_BCLK, AUDIO_I2S_SPK_GPIO_LRCK, AUDIO_I2S_SPK_GPIO_DOUT, AUDIO_I2S_MIC_GPIO_SCK, AUDIO_I2S_MIC_GPIO_WS, AUDIO_I2S_MIC_GPIO_DIN);
      |             ^~~~~~~~~~~~~~~~~~~~~~~
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/config.h:13:33: error: 'GPIO_NUM_41' was not declared in this scope; did you mean 'GPIO_NUM_4'?
   13 | #define AUDIO_I2S_MIC_GPIO_SCK  GPIO_NUM_41
      |                                 ^~~~~~~~~~~
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc:123:88: note: in expansion of macro 'AUDIO_I2S_MIC_GPIO_SCK'
  123 |             AUDIO_I2S_SPK_GPIO_BCLK, AUDIO_I2S_SPK_GPIO_LRCK, AUDIO_I2S_SPK_GPIO_DOUT, AUDIO_I2S_MIC_GPIO_SCK, AUDIO_I2S_MIC_GPIO_WS, AUDIO_I2S_MIC_GPIO_DIN);
      |                                                                                        ^~~~~~~~~~~~~~~~~~~~~~
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/config.h:12:33: error: 'GPIO_NUM_42' was not declared in this scope; did you mean 'GPIO_NUM_4'?
   12 | #define AUDIO_I2S_MIC_GPIO_WS   GPIO_NUM_42
      |                                 ^~~~~~~~~~~
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc:123:112: note: in expansion of macro 'AUDIO_I2S_MIC_GPIO_WS'
  123 |             AUDIO_I2S_SPK_GPIO_BCLK, AUDIO_I2S_SPK_GPIO_LRCK, AUDIO_I2S_SPK_GPIO_DOUT, AUDIO_I2S_MIC_GPIO_SCK, AUDIO_I2S_MIC_GPIO_WS, AUDIO_I2S_MIC_GPIO_DIN);
      |                                                                                                                ^~~~~~~~~~~~~~~~~~~~~
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc: In member function 'virtual Backlight* KEVIN_SP_V3Board::GetBacklight()':
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/config.h:39:31: error: 'GPIO_NUM_48' was not declared in this scope; did you mean 'GPIO_NUM_8'?
   39 | #define DISPLAY_BACKLIGHT_PIN GPIO_NUM_48
      |                               ^~~~~~~~~~~
/Users/jintinghou/tmp/GitHub/xiaozhi-esp32/main/boards/kevin-sp-v3-dev/kevin-sp-v3_board.cc:132:39: note: in expansion of macro 'DISPLAY_BACKLIGHT_PIN'
  132 |         static PwmBacklight backlight(DISPLAY_BACKLIGHT_PIN, DISPLAY_BACKLIGHT_OUTPUT_INVERT);
      |        

### Steps to reproduce.

idf.py set-target esp32c3
idf.py build

### Build or installation Logs.

```plain

```

78, 2025-04-15:
这个板子不应该用 `idf.py set-target esp32s3` 吗？

jthou, 2025-04-16:
应该跟板子没关系，跟芯片型号有关系。选别的板子也是这样错误（我是个新手，还没弄清楚板子和芯片型号等组合关系，）

Kevincoooool, 2025-04-16:
就是和板子有关系，你自己的板子什么型号就要选对应的

aeolus21, 2025-04-18:
请问问题解决了吗，我也碰到了，esp32c3

Kevincoooool, 2025-04-18:
> 请问问题解决了吗，我也碰到了，esp32c3

自己的板子是C3就要选对应board型号为C3的开发板，并且此前应该idf.py set-target esp32c3

jthou, 2025-04-19:
板子选了 虾哥 c3，再编译就好了。中间遇到过 uint_8 没有定义的问题，后来又消失了。

jthou, 2025-04-19:
总结：
1，esp-idf 中，执行./install.sh 的时候，网络一定要好，要是中途断了，要把~/.espressif 目录删掉重新执行；2, 执行 . .export.sh ；3. 执行 idf.py menuconfig 选对板子（虽然我的板子还没到）；4，执行 idf.py set-target esp32c3; 5. 执行 idf.py build