78/xiaozhi-esp32 #504: idf.py build FAILED: xiaozhi.elf

104gogo, 2025-04-17:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.6.0-1-ga6619dc

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

期望能正常build

### What is the actual behavior?

新 clone 的项目，在 main 分支使用 idf.py build 后出现了 FAILED: xiaozhi.elf 的错误信息

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain
-- Configuring done (5.0s)
-- Generating done (0.1s)
-- Build files have been written to: C:/Users/gogo/Programs/github/test/xiaozhi-esp32/build/bootloader
[1787/1924] Building C object esp-idf/espres...sif__esp_lcd_gc9a01.dir/esp_lcd_gc9a01.c.ob
In file included from C:/Users/gogo/Programs/github/test/xiaozhi-esp32/managed_components/espressif__esp_lcd_gc9a01/include/esp_lcd_gc9a01.h:13,
                 from C:/Users/gogo/Programs/github/test/xiaozhi-esp32/managed_components/espressif__esp_lcd_gc9a01/esp_lcd_gc9a01.c:20:
C:/Espressif/frameworks/esp-idf-v5.4/components/esp_hw_support/include/intr_types.h:8:2: warning: #warning "This header is deprecated. Please use esp_intr_types.h instead" [-Wcpp]
    8 | #warning "This header is deprecated. Please use esp_intr_types.h instead"
      |  ^~~~~~~
[114/115] Generating binary image from built executable
esptool.py v4.8.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated C:/Users/gogo/Programs/github/test/xiaozhi-esp32/build/bootloader/bootloader.bin
[115/115] C:\WINDOWS\system32\cmd.exe /C "cd...aozhi-esp32/build/bootloader/bootloader.bin
Bootloader binary size 0x4040 bytes. 0x3fc0 bytes (50%) free.
[1922/1924] Linking CXX executable xiaozhi.elf
FAILED: xiaozhi.elf
C:\WINDOWS\system32\cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp-elf\esp-14.2.0_20241119\xtensa-esp-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=C:/Users/gogo/Programs/github/test/xiaozhi-esp32/build/xiaozhi.map -Wl,--no-warn-rwx-segments -Wl,--orphan-handling=warn -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.bt_funcs.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.wdt.ld -T esp32s3.rom.version.ld -T esp32s3.rom.newlib.ld -T memory.ld -T sections.ld @CMakeFiles\xiaozhi.elf.rsp -o xiaozhi.elf && cd ."
C:/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/gogo/Programs/github/test/xiaozhi-esp32/managed_components/espressif__esp-sr/lib/esp32s3/libc_speech_features.a(esp_speech_features.c.obj):(.literal.esp_fftr+0xc): undefined reference to `dsps_fft4r_fc32_ae32_'
C:/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/gogo/Programs/github/test/xiaozhi-esp32/managed_components/espressif__esp-sr/lib/esp32s3/libc_speech_features.a(esp_speech_features.c.obj): in function `esp_fftr':
/home/sunxiangyu/workspace/esp_sr_lib/components/c_speech_features/esp_speech_features/esp_speech_features.c:294:(.text.esp_fftr+0x31): undefined reference to `dsps_fft4r_fc32_ae32_'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\gogo\Programs\github\test\xiaozhi-esp32\build\log\idf_py_stderr_output_20156 and C:\Users\gogo\Programs\github\test\xiaozhi-esp32\build\log\idf_py_stdout_output_20156
```

eastack, 2025-04-17:
Haha, I just ran into the same issue — did you manage to fix it?

eastack, 2025-04-17:
It appears to be related to some updates in esp-dsp version 1.6.0 https://github.com/espressif/esp-sr/issues/140

eastack, 2025-04-17:
Manually specifying the esp-dsp dependency and its version under dependencies in main/idf_component.yml at the project root directory can resolve this issue.
[image]

104gogo, 2025-04-17:
Yes, thank you!

closed 2025-04-17 by 104gogo

manicwy, 2025-04-17:
> Manually specifying the esp-dsp dependency and its version under dependencies in main/idf_component.yml at the project root directory can resolve this issue. [image]

奇怪，我添加了这个依赖项目，重新编译还是问题依旧

sun-xiangyu, 2025-04-17:
> > Manually specifying the esp-dsp dependency and its version under dependencies in main/idf_component.yml at the project root directory can resolve this issue. [image]
> 
> 奇怪，我添加了这个依赖项目，重新编译还是问题依旧

可以尝试将esp-sr 的版本固定在 2.0.3， 比如
espressif/esp-sr: "2.0.3"

tangyumei3535, 2025-04-17:
esp-sr bug 修了，改成 espressif/esp-sr: "2.0.6" 可以了