78/xiaozhi-esp32 #506: [4/7] Linking CXX executable xiaozhi.elf FAILED: xiaozhi.elf

manicwy, 2025-04-17:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.6.0

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Build successful

### What is the actual behavior?

[4/7] Linking CXX executable xiaozhi.elf
FAILED: xiaozhi.elf
C:\windows\system32\cmd.exe /C "cd . && E:\Espressif\tools\xtensa-esp-elf\esp-14.2.0_20241119\xtensa-esp-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=F:/xiaozhi-esp32/build/xiaozhi.map -Wl,--no-warn-rwx-segments -Wl,--orphan-handling=warn -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.bt_funcs.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.wdt.ld -T esp32s3.rom.version.ld -T esp32s3.rom.newlib.ld -T memory.ld -T sections.ld @CMakeFiles\xiaozhi.elf.rsp -o xiaozhi.elf && cd ."
E:/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: F:/xiaozhi-esp32/managed_components/espressif__esp-sr/lib/esp32s3/libc_speech_features.a(esp_speech_features.c.obj):(.literal.esp_fftr+0xc): undefined reference to dsps_fft4r_fc32_ae32_' E:/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: F:/xiaozhi-esp32/managed_components/espressif__esp-sr/lib/esp32s3/libc_speech_features.a(esp_speech_features.c.obj): in function esp_fftr':
/home/sunxiangyu/workspace/esp_sr_lib/components/c_speech_features/esp_speech_features/esp_speech_features.c:294:(.text.esp_fftr+0x31): undefined reference to `dsps_fft4r_fc32_ae32_'

collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

manicwy, 2025-04-17:
好像espressif/idf-dsp昨天更新到1.6.0版本后才出现这个编译链接问题

FreezeLee1, 2025-04-17:
> 快速使用方法是先将 esp-sr 里面的 esp-dsp 版本 固定为 1.5.2，等他们修 bug

 请教 espressif/esp-dsp: "1.5.2" 固定了版本也不行，是哪里还有问题

sun-xiangyu, 2025-04-17:
很抱歉这次由于 esp-dsp 升级产生的事故，现在的解决方法
esp-sr v1.* 升级到 esp-sr v1.9.5
esp-sr v2.* 升级到 esp-sr v2.0.5 或更新

这次错误产生的原因是 由于 esp-dsp 在 v1.6.0 版本中的fft4r 函数变动，导致esp-sr库无法正确链接到fft4r目标函数。
在 esp-sr v2.0.5 及以后都是基于新的函数编译，需要使用 esp-dsp v1.6.0
在 esp-sr v2.0.5 以前的版本，需要使用 esp-dsp v1.5.2 或 更早的版本

manicwy, 2025-04-17:
> 很抱歉这次由于 esp-dsp 升级产生的事故，现在的解决方法 esp-sr v1.* 升级到 esp-sr v1.9.5 esp-sr v2.* 升级到 esp-sr v2.0.5 或更新
> 
> 这次错误产生的原因是 由于 esp-dsp 在 v1.6.0 版本中的fft4r 函数变动，导致esp-sr库无法正确链接到fft4r目标函数。 在 esp-sr v2.0.5 及以后都是基于新的函数编译，需要使用 esp-dsp v1.6.0 在 esp-sr v2.0.5 以前的版本，需要使用 esp-dsp v1.5.2 或 更早的版本

好像没这个2.0.5版本，

-idf-v5.4/tools/cmake/build.cmake:610 (message):
  WARNING: Component "espressif/esp-sr" not found

  ERROR: Because project depends on espressif/esp-sr (2.0.5) which doesn't
  match any versions, version solving failed.

> 很抱歉这次由于 esp-dsp 升级产生的事故，现在的解决方法 esp-sr v1.* 升级到 esp-sr v1.9.5 esp-sr v2.* 升级到 esp-sr v2.0.5 或更新
> 
> 这次错误产生的原因是 由于 esp-dsp 在 v1.6.0 版本中的fft4r 函数变动，导致esp-sr库无法正确链接到fft4r目标函数。 在 esp-sr v2.0.5 及以后都是基于新的函数编译，需要使用 esp-dsp v1.6.0 在 esp-sr v2.0.5 以前的版本，需要使用 esp-dsp v1.5.2 或 更早的版本

把版本固定在 2.0.3 可以build成功

DOVE-PH, 2025-04-18:
有v2.0.5的，改了这个就能编译了

78, 2025-05-20:
> > 快速使用方法是先将 esp-sr 里面的 esp-dsp 版本 固定为 1.5.2，等他们修 bug
> 
> 如何进行升级

该问题在新版本代码已经不存在。

ruanrongman, 2025-05-21:
> > > 快速使用方法是先将 esp-sr 里面的 esp-dsp 版本 固定为 1.5.2，等他们修 bug
> > 
> > 
> > 如何进行升级
> 
> 该问题在新版本代码已经不存在。

还是在报错，这个问题在idf5.3.3编译以及5.3.2编译均出现link问题

sun-xiangyu, 2025-05-21:
> 
> 还是在报错，这个问题在idf5.3.3编译以及5.3.2编译均出现link问题

建议删除缓存的版本，缓存版本一般存放在项目目录下的 `managed_components` ,找到这个目录删除，就会拉取最新的版本