78/xiaozhi-esp32 #516: 请注意mqtt.xiaozhi.me服务端安全策略保护

ChenJBBB, 2025-04-18:
服务端连接信息通过如下代码
``` c
    Settings settings("mqtt", false);
    endpoint_ = settings.GetString("endpoint");
    client_id_ = settings.GetString("client_id");
    username_ = settings.GetString("username");
    password_ = settings.GetString("password");
    publish_topic_ = settings.GetString("publish_topic");
    // 打印连接信息
    ESP_LOGI(TAG, "MQTT endpoint: %s, client_id: %s, username: %s, password: %s, publish_topic: %s", endpoint_.c_str(), client_id_.c_str(), username_.c_str(), password_.c_str(), publish_topic_.c_str());
```

可以打印连接信息。 连接信息在MQTT客户端软件是可以直接连接上使用的，且可以正常发布消息。 可能对只主题做了部分权限鉴权。要注意防范攻击~~

[image]

[image]

78, 2025-04-18:
可能会有哪一类的攻击？

ChenJBBB, 2025-04-18:
> 可能会有哪一类的攻击？

使用Qos2 遍历主题订阅. 消耗服务器网络资源（当然对攻击者自己也是一种消耗，但是有心人可能肉鸡多）。
如果对主题有权限控制能缓解一大部分

opcache, 2025-04-21:
要做 IP白名单限制，或者上WAF，WAF可以看下这个 https://github.com/samwafgo/SamWaf
我觉得不错

78, 2025-04-21:
> > 可能会有哪一类的攻击？
> 
> 使用Qos2 遍历主题订阅. 消耗服务器网络资源（当然对攻击者自己也是一种消耗，但是有心人可能肉鸡多）。 如果对主题有权限控制能缓解一大部分

有考虑，目前只支持 Qos 0

closed 2025-04-23 by ChenJBBB