78/xiaozhi-esp32 #527: 使用1.6版本后，esp box lite出现破音现象

chencharry, 2025-04-22:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.6.0-4-gabb594b

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

不清楚为何会有破音。硬件无改动。

### What is the actual behavior?

esp box lite出现破音现象

### Steps to reproduce.

上电对话之后都有这个问题

### Build or installation Logs.

```plain

```

chencharry, 2025-04-22:
实测1.5.8已经有这个问题。

78, 2025-04-22:
好像是因为合并了一个开了aec功能的版本，你看看是否可以关闭？

mtdxc, 2025-04-30:
espbox_lite可以在编译选项中手工将aec关掉。

chencharry, 2025-05-05:
> espbox_lite可以在编译选项中手工将aec关掉，跟开关微信聊天样式一样。

具体是哪一个呢？如果是“ Enable C++ run-time type info (RTTI)要 AEC 支持）”的话，这个没有打开。
目前看到的是只有“CONFIG_AFE_INTERFACE_V1”是y的。但是好像不能取消。

mtdxc, 2025-05-06:
CONFIG_USE_AUDIO_PROCESSOR

mtdxc, 2025-05-06:
[image]

chencharry, 2025-05-06:
> [image]

默认这个是不勾选的。我把它勾选上。确实好用了。谢谢。

closed 2025-05-06 by chencharry