78/xiaozhi-esp32 #530: 从OTA接口读取Websocket服务器后如何自定义后端服务器地址？

neil0306, 2025-04-22:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.6.1

### Operating System used.

Linux

### How did you build your project?

VS Code IDE/Cursor

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

[image]
拉取了最新的commit之后，发现删除了自定义 websocket URL 的配置， 想问一下最新版代码应该如何配置自定义后端服务器的URL呢？

### What is the actual behavior?

删除了自定义websocket URL 的配置项

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

78, 2025-04-22:
如果你使用的是 xiaozhi-esp32-server 部署自定义后端服务器，目前已经支持在 OTA 接口返回 websocket 连接信息：https://github.com/xinnan-tech/xiaozhi-esp32-server/pull/925

Jimmy-L99, 2025-04-23:
> 如果你使用的是 xiaozhi-esp32-server 部署自定义后端服务器，目前已经支持在 OTA 接口返回 websocket 连接信息：[xinnan-tech/xiaozhi-esp32-server#925](https://github.com/xinnan-tech/xiaozhi-esp32-server/pull/925)

请教一下，我用的是最简化源码部署，没有智控台，但用的是最新版的xiaozhi-esp32，里面已经没有自定义websocket了，那该在哪设置？

neil0306, 2025-04-23:
> > 如果你使用的是 xiaozhi-esp32-server 部署自定义后端服务器，目前已经支持在 OTA 接口返回 websocket 连接信息：[xinnan-tech/xiaozhi-esp32-server#925](https://github.com/xinnan-tech/xiaozhi-esp32-server/pull/925)
> 
> 请教一下，我用的是最简化源码部署，没有智控台，但用的是最新版的xiaozhi-esp32，里面已经没有自定义websocket了，那该在哪设置？

根据 xiaozhi-esp32-server 的开发者反馈，目前需要【全功能部署】才能使用，只部署server端，没有部署管理模块的话无法使用这个功能，也就是说小智固件只能退回1.6.0版本。

78, 2025-04-23:
原来还有这样的问题，确实需要考虑，目前还没发布固件，可以看看有什么好的解决方案。

bao17634, 2025-04-23:
> 根据 xiaozhi-esp32-server 的开发者反馈，目前需要【全功能部署】才能使用，只部署server端，没有部署管理模块的话无法使用这个功能，也就是说小智固件只能退回1.6.0版本。

我已经部署了全功能了，因为我之前已经烧录过小智固件了，导致每次烧录不会重新提供激活码。导致也连不上我自己的管理后台

openrz, 2025-04-24:
[全模块部署的教程](https://github.com/xinnan-tech/xiaozhi-esp32-server/blob/main/docs/firmware-build.md)已经更新了，试试看

[image]

bao17634, 2025-04-24:
> [全模块部署的教程](https://github.com/xinnan-tech/xiaozhi-esp32-server/blob/main/docs/firmware-build.md)已经更新了，试试看
> 
> [image]
感谢 大佬 神速啊

liu731, 2025-04-27:
请教，看Commit已经删除了CONNECTION_TYPE，那么怎么才能继续使用Websocke方式连接xiaozhi.me(https://api.tenclass.net/xiaozhi/ota/)? 默认返回的是MQTT相关信息。

104gogo, 2025-04-27:
不想在本地运行全量xiaozhi-esp32-server

> 原来还有这样的问题，确实需要考虑，目前还没发布固件，可以看看有什么好的解决方案。

https://xiaozhi.me/ 加个 ws url 配置的功能吧

78, 2025-04-27:
> 不想在本地运行全量xiaozhi-esp32-server
> 
> > 原来还有这样的问题，确实需要考虑，目前还没发布固件，可以看看有什么好的解决方案。
> 
> https://xiaozhi.me/ 加个 ws url 配置的功能吧

我觉得你在本地目录写个 JSON 文件然后 `python -m http.server 80` Host 这个文件，作为 OTA 接口提供给固件更快捷。

OTA接口的返回值参考：https://ccnphfhqs21z.feishu.cn/wiki/FjW6wZmisimNBBkov6OcmfvknVd

104gogo, 2025-04-28:
> > 不想在本地运行全量xiaozhi-esp32-server
> > > 原来还有这样的问题，确实需要考虑，目前还没发布固件，可以看看有什么好的解决方案。
> > 
> > 
> > https://xiaozhi.me/ 加个 ws url 配置的功能吧
> 
> 我觉得你在本地目录写个 JSON 文件然后 `python -m http.server 80` Host 这个文件，作为 OTA 接口提供给固件更快捷。
> 
> OTA接口的返回值参考：https://ccnphfhqs21z.feishu.cn/wiki/FjW6wZmisimNBBkov6OcmfvknVd

也可以，谢谢

tiann, 2025-05-01:
> > 如果你使用的是 xiaozhi-esp32-server 部署自定义后端服务器，目前已经支持在 OTA 接口返回 websocket 连接信息：[xinnan-tech/xiaozhi-esp32-server#925](https://github.com/xinnan-tech/xiaozhi-esp32-server/pull/925)
> 
> 请教一下，我用的是最简化源码部署，没有智控台，但用的是最新版的xiaozhi-esp32，里面已经没有自定义websocket了，那该在哪设置？

目前简化版的部署也支持 ota 返回了，这个 issue 应该可以关闭掉。

ref: https://github.com/xinnan-tech/xiaozhi-esp32-server/releases/tag/v0.3.12

closed 2025-05-01 by 78