78/xiaozhi-esp32 #535: 如何设置自己搭建的后台服务

bao17634, 2025-04-23:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.6.0

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

但用的是最新版的xiaozhi-esp32固件编译的固件，里面我`
config OTA_URL
    string "Default OTA URL"
    # default "https://api.tenclass.net/xiaozhi/ota/"
    default "http://localhost:8002/xiaozhi/ota/"
    help
        The application will access this URL to check for new firmwares and server address.` 这个地方我配置了我自己搭建的服务，但是在烧录后不生效

### What is the actual behavior?

有没有哪个地方可以配置自己的搭建的后台服务

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

RayforLoy, 2025-04-24:
新版直接从ota server获取协议地址，要自己改一下protocol里面的几个文件

loadingvx, 2025-04-24:
1.6.0 非常奇怪的改动. 

[image]

此代码改动是为了提高开发者的门槛吗,或者屏蔽三方服务器吗? 为什么删掉配置项中的三方服务器地址参数?

bao17634, 2025-04-24:
> 1.6.0 非常奇怪的改动.
> 
> [image]
> 
> 此代码改动是为了提高开发者的门槛吗,或者屏蔽三方服务器吗? 为什么删掉配置项中的三方服务器地址参数?

是为了方便服务端的配置把，无需编译就可以更新地址，现在新的代码已经修复了这个问题

78, 2025-04-24:
新版本的 esp-wifi-connect 已经支持配置 OTA 地址并保存到 NVS flash。
可以无需编译固件，即可通过更换 OTA 地址来切换不同的后端服务器。

loadingvx, 2025-04-24:
好的, 不过现在挺困扰的, 开发板只要启动就会更新固件, 还是需要重新编译. 
看来还是要尽早实现自己的ota接口.

78, 2025-04-24:
> 好的, 不过现在挺困扰的, 开发板只要启动就会更新固件, 还是需要重新编译. 看来还是要尽早实现自己的ota接口.

目前开源 server 已经有了 ota 接口的支持，可以同步到最新的代码。