78/xiaozhi-esp32 #536: Different Emojis please. Also Top status bar in English please.

orion14794, 2025-04-24:
### Is your feature request related to a problem?

Greetings. I have 2 xingzhi-cube-1.54 tft-wifi units. Found the following website: https://ccnphfhqs21z.feishu.cn/wiki/W14Kw1s1uieoKjkP8N0c1VVvn8d which offers a firmware version 1.60 for these Ai devices. 1 of my units was running version 1.46 and the other was on 1.54. I was wondering why both devices had different versions. I bought them both on Aliexpress. So after looking around the internet found the website posted on this post, flashed the devices with the firmware and got both devices updated to 1.60. My only issue is the top status bar text is in Chinese characters, but the text from the Ai typed is English. Also lost the emojis that which were 3 different one than the stock yellow face. Also both devices receive OTA updates and a Dark mode theme has been added. Is there going to be a versions full English and with other kinds of Emojis? Admires the work you've done so far and these ESP 32 devices are amazing! Cheers.

likeaturtle, 2025-05-04:
> > At present, if you wish to switch to English or Japanese, you need to compile the code using VSCODE and ESP-IDF 5.3. There is a possibility that in subsequent versions, we will incorporate support for various languages within a single firmware.目前，如果您想切换到英语或日语，您需要使用 VSCODE 和 ESP-IDF 5.3 编译代码。有可能在后续版本中，我们将支持在单个固件中集成多种语言。
> 
> Hello and thanks for the reply. I'm not a programmer or coder and don't know how to compile stuff. I'm just a dude that purchased the devices on Aliexpress and that's it. I managed to update the firmware from reading a post about flashing the firmwares and just followed what was instructed to do. But regardless, my devices are working well, just as I mentioned the top status bar is Chinese characters, while the Ai's responses are English since I picked the Ai voice options English in the Dashboard. One of the things I like about the latest firmware is the Theme change that can be changed from White to Dark by prompting the command using voice. Also that the devices can receive over the air updates to update the firmware. You guys are fantastic making these things work and creating the firmware! Super smart! Thanks for such awesome ideas and for sharing them. Sending best wishes and blessings from the USA.问候并感谢您的回复。我不是程序员或编码者，不知道如何编译东西。我只是个在 Aliexpress 上购买设备的普通人，仅此而已。我通过阅读有关刷固件的文章，成功更新了固件，只是按照指示做了。但不管怎样，我的设备工作得很好，就像我说的，顶部状态栏是中文字符，而 AI 的回答是英文，因为我已经在仪表板上选择了 AI 语音选项为英文。我喜欢最新固件的一个特点是可以通过语音提示命令将主题从白色改为黑色。还有设备可以通过空中更新来更新固件。你们让这些设备工作得如此出色，并创建了固件！太聪明了！感谢您有如此棒的想法，并与大家分享。从美国向您送上最美好的祝愿和祝福。

If you wish, I can compile it for you in English. Please provide your device model, the desired wake word and the email address for receiving the firmware.

Opsy1337, 2025-07-08:
> If you wish, I can compile it for you in English. Please provide your device model, the desired wake word and the email address for receiving the firmware.

Dear,
Kindly I need your support to change the language in screen to English, also I want change Emojis to custom one, if you can help me send me mail please

**ixbiat@gmail.com**