78/xiaozhi-esp32 #542: OTA: No serial number found

troey11, 2025-04-25:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

v1.6

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

[image]
I tried to install it on esp32 wroom, there was an error.

i use firmware v1.6.0_bread-compact-esp32

### What is the actual behavior?

.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

78, 2025-04-25:
This does not mean an error. Kindly add the device on the control panel: xiaozhi.me

troey11, 2025-04-25:
> This does not mean an error. Kindly add the device on the control panel: xiaozhi.me

i deploy this only esp32 board without mic,speaker and lcd. 
I have tried to add the device. How to get verification code, if without lcd and speaker ?

78, 2025-04-25:
The UART console will print the verification code.

troey11, 2025-04-29:
> The UART console will print the verification code.

wow, thanks

closed 2025-04-29 by troey11