78/xiaozhi-esp32 #544: 声音卡顿,消失,无法播放

loadingvx, 2025-04-25:
### Answers checklist.

- [x] I have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI version.

>= 1.5.9 

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

esp32-s3 开发板, 语音正常播放

### What is the actual behavior?

1.5.8 可以正常播放语音, 1.5.9版本只有, 播放声音出现问题, 总是会卡在第一个字 ,然后后面偶尔有部分声音, 无法正常播放声音. 

### Steps to reproduce.

server端使用的是edge-tts, 采样未做任何改变都是16000, 单声道, 60ms.

### Build or installation Logs.

```plain

```

loadingvx, 2025-04-25:
回滚编译1.5.8之后, 恢复正常. 是否1.5.9只有修改了音频相关代码?  server端如何适配?

findstrbot, 2025-04-25:
我也碰到这个问题了，本质上是因为最新的小智帧缓冲只有5， 所以依赖于一个高精度的服务端定时器，不然流控极难把握，我已经提了[issue](https://github.com/78/xiaozhi-esp32/issues/545)不知道官方会不会加。

loadingvx, 2025-05-04:
> 我也碰到这个问题了，本质上是因为最新的小智帧缓冲只有5， 所以依赖于一个高精度的服务端定时器，不然流控极难把握，我已经提了[issue](https://github.com/78/xiaozhi-esp32/issues/545)不知道官方会不会加。

非常正确, 果然主要是server端的问题, 现在已经解决了. 感谢提供方向 !

closed 2025-05-07 by 78

littlerookie, 2025-05-14:
> > 我也碰到这个问题了，本质上是因为最新的小智帧缓冲只有5， 所以依赖于一个高精度的服务端定时器，不然流控极难把握，我已经提了[issue](https://github.com/78/xiaozhi-esp32/issues/545)不知道官方会不会加。
> 
> 非常正确, 果然主要是server端的问题, 现在已经解决了. 感谢提供方向 !

方便说下怎么解决的嘛？