78/xiaozhi-esp32 #545: 小智能否在握手时增加协商opus帧缓冲大小

findstrbot, 2025-04-25:
### Is your feature request related to a problem?

我正在编写一个[运行在树莓派或Nas上的小型Server, 目标内存是100M以内](https://github.com/findstr/xiaozhi-esp32-server-mini)。

最新的小智opus帧缓冲只有5帧，一帧是60ms, 5帧只有300ms

由于我自己的服务器框架精度(25ms)问题，流控极难把握，不小心就会发超或者不连续，造成读音太大。

因此，小智官方是否可以在握手时，由服务器来提供一个opus的帧缓冲大小，如果过大（比如100）则直接拒绝。

这样适配性应该会更广一些。

如果官方接受，我可以来实现相关的PR.

78, 2025-04-25:
服务器端应该进行流控，以下是参考代码：

```javascript
class RateController extends EventEmitter {
  constructor(sampleRate, frameDuration) {
    super();
    this.sampleRate = sampleRate || DEFAULT_SAMPLE_RATE;
    this.frameDuration = frameDuration || DEFAULT_FRAME_DURATION;
    this.queue = [];
    this.reset();
  }

  reset() {
    if (this.rateControlTimeout) {
      clearTimeout(this.rateControlTimeout);
      this.rateControlTimeout = null;
    }
    if (this.isBusy()) {
      // 如果已发送 start 事件，或者 stop 事件在队列中，则需要发送 stop 事件
      this.queue = [];
      this.sendStop();
    }

    this.encoder = new OpusEncoder(this.sampleRate, 1);
    this.encoder.applyEncoderCTL(4000, 2048); // VOIP
    // this.encoder.applyEncoderCTL(4010, 5); // Complexity
    this.encodeFrameSize = this.sampleRate / 1000 * this.frameDuration;
    this.audioBuffer = Buffer.alloc(0);
    this.playPosition = -this.frameDuration * 2; // 初始化播放位置
    this.startTimestamp = null;
    this.started = false;
  }

  checkQueue() {
    while (this.queue.length > 0) {
      // 检查队列中的第一项是否为 JSON 对象或 PCM 数据
      if (this.queue[0] instanceof Buffer) {
        // 检查是否是时候发送音频
        if (this.rateControlTimeout) {
          break;
        }
        if (this.startTimestamp === null) {
          this.startTimestamp = Date.now();
        }
        const elapsedMs = Date.now() - this.startTimestamp;
        const outputMs = this.playPosition;
        if (elapsedMs < outputMs) {
          this.rateControlTimeout = setTimeout(() => {
            this.rateControlTimeout = null;
            this.checkQueue();
          }, outputMs - elapsedMs);
          break;
        }
        const pcm = this.queue.shift();
        // 在这里进行 Opus 编码，而不是直接从队列中获取已编码的数据
        const opus = this.encoder.encode(pcm);
        this.playPosition += this.frameDuration;
        this.emit('audio', opus);
      } else {
        this.emit('message', this.queue.shift());
      }
    }
  }

  sendAudio(pcm) {
    this.audioBuffer = Buffer.concat([this.audioBuffer, pcm]);
    const pieceLength = this.encodeFrameSize * 2;
    while (this.audioBuffer.length >= pieceLength) {
      const frame = this.audioBuffer.subarray(0, pieceLength);
      this.audioBuffer = this.audioBuffer.subarray(pieceLength);
      // 直接将 PCM 帧添加到队列，不进行编码
      this.queue.push(frame);
    }
    this.checkQueue();
  }

  sendSentenceStart(text) {
    this.queue.push({ type: 'tts', state: 'sentence_start', text });
    this.checkQueue();
  }

  sendSentenceEnd(text) {
    this.queue.push({ type: 'tts', state: 'sentence_end', text });
    this.checkQueue();
  }

  sendStart() {
    this.queue.push({ type: 'tts', state: 'start', sample_rate: this.sampleRate });
    this.started = true;
    this.checkQueue();
  }

  sendStop() {
    this.queue.push({ type: 'tts', state: 'stop' });
    this.started = false;
    this.checkQueue();
  }

  isBusy() {
    return this.started || this.queue.length > 0;
  }
}
```

findstrbot, 2025-04-27:
好的，谢谢

closed 2025-04-27 by findstrbot