78/xiaozhi-esp32 #554: 请问1.6.0里，26个免费唤醒词，没了么？

xuyl-china-soft, 2025-04-27:
### Answers checklist.

- [x] I  have read the documentation [XiaoZhi AI Programming Guide](https://ccnphfhqs21z.feishu.cn/wiki/F5krwD16viZoF0kKkvDcrZNYnhb) and the issue is not addressed there.
- [x] I have updated my firmware to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### XiaoZhi AI firmware version.

bread-compact-wifi-lcd

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

CMD

### Power Supply used.

Battery

### What is the expected behavior?

想使用其它免费的唤醒词

### What is the actual behavior?

请问1.6.0里，26个免费唤醒词，已经没了么？
我在自己环境里，menuconfig下已经找不到，相关下拉选项了。

### Steps to reproduce.

1. 进入menuconfig
2. 找到ESP Speech Recognition
3. 免费提示词下拉不见了。
早期版本里有，但不确定那一版开始没了。

### Debug Logs.

```plain

```

likeaturtle, 2025-04-27:
似乎是 vscode 的 idf 插件问题，用命令行是可以显示的

78, 2025-04-27:
试试切换以下 esp-sr 的版本 比如改成 =="2.0.3"

yezhihao312, 2025-04-28:
请问在哪可以改esp-sr 的版本

78, 2025-04-28:
> 请问在哪可以改esp-sr 的版本

main下的 idf_component.yml

kgdsqq520, 2025-04-28:
> > 请问在哪可以修改 esp-sr 的版本
> 
> 主下的 idf_component.yml

没有用，从2.0.5改为2.0.3之后，依然没有唤醒词，但改为1.9.0就出现唤醒词了，但是会构建失败。拉取1.6.0版本也是这样。

78, 2025-04-29:
你的 esp-idf 是什么版本？ 我用的是 5.4  的版本，最新版本的仓库代码，是可以选择提示词的。

[image]

CCFQ, 2025-04-29:
[image]
你好我也是5.4的版本但也没有显示。代码是今天早上11点前后拉取的

jvssoft, 2025-04-29:
espressif/esp-sr: "==2.0.3"这个可以

CCFQ, 2025-04-29:
@jvssoft 谢谢，这个是有用的

yezhihao312, 2025-04-29:
按照上面说的，在main下的 idf_component.yml，把esp-sr 的版本改成 “==2.0.3”，唤醒词就出来了，完美！我的小冰回来了

yuelanhu7, 2025-05-05:
> > > 请问在哪可以修改 esp-sr 的版本
> > 
> > 
> > 主下的idf_component.yml
> 
> 没有用，从2.0.5改为2.0.3之后，依然没有唤醒词，但改为1.9.0就出现唤醒词了，但是会构建失败。拉取1.6.0版本也是这样。

我的也是一样